KMKfw/kmk_firmware #528: Chord doesn't always respect timeout value

Jawfish, 2022-07-21:
The chord feature of the combos module doesn't use a user-set timeout until that timeout has passed since the last keystroke.
Example:
```python
keyboard.modules = [
    combos,
]
combos.combos = [Chord((KC["1"], KC["2"]), KC["3"], timeout=5000)]
keyboard.keymap = [K['1'],  K['2'],  K['3']]
```
Typing 1 or 2 within the first 5000 milliseconds of the previous non-chord keypress will send 1 or 2 rather than waiting up to 5000ms for the next keypress. The chord still works if you press it within the default 50ms timeout. After 5000ms has passed since the last keystroke, it works as expected.

Current behavior:
1. B is tapped
2. 1 is held
3. 1 is sent by the keyboard to the computer
4. 400ms passes
5. 2 is tapped
6. 2 is sent by the keyboard to the computer
7. 5000ms passes
8. 1 is held
9. 400ms passes
10. 2 is tapped
11. 3 is sent by the keyboard to the computer

Expected behavior:
1. B is tapped
2. 1 is held
3. 400ms passes
4. 2 is tapped
5. 3 is sent by the keyboard to the computer

xs5871, 2022-07-21:
That's "intentional". You pressed B+1+2 within the combo timeout, and that key combination doesn't match the chord 1+2.
Is what you expect a rolling matching window?

Jawfish, 2022-07-21:
B is released before 1 is pressed in my example.

Jawfish, 2022-07-21:
I updated my comment to make it more clear, previously I just said 'tap' to mean a press-and-release action.

Jawfish, 2022-07-24:
Here's a test I wrote for this that fails:

```python
# the config
combos.combos = [
    Chord((KC.F, KC.G), KC.Z),
]
self.keyboard = KeyboardTest(
    [combos, layers],
    [
        [KC.A, KC.B, KC.C, KC.D, KC.E, KCMO, KC.F, KC.G],
        [KC.N1, KC.N2, KC.N3, KC.N4, KC.N5, KC.LEADER],
    ],
)

# the test
keyboard.test(
    'match: Other pressed before combo',
    [
        (1, True),
        (1, False),
        (7, True),
        (6, True),
        (6, False),
        (7, False),
        t_after,
    ],
    [{KC.B}, {}, {KC.Z}, {}],
)
```
The failure:
```python
assert mods: 0 == 0 keys: {'B'} == {'B'} 
assert mods: 0 == 0 keys: set() == set() 
assert mods: 0 == 0 keys: {'G'} == {'Z'} 
assert mods: 0 == 0 keys: {'F', 'G'} == set() 
assert mods: 0 == None keys: {'G'} == {None} 
assert mods: 0 == None keys: set() == {None} 
```
Remove the `KC.B` press and it passes. I'll look into the code tomorrow if I have time.

closed 2022-07-27 by xs5871