KMKfw/kmk_firmware #534: Mouse keys not working

electricWah, 2022-07-25:
Some new features have been added since I last download kmk, but when I downloaded the newest version, mousekeys stopped working. When I try to move the mouse, it does move a tiny bit but then it stops. I switched back to the old version of `kmk/modules/mouse_keys.py` and it works again, so I assume that's where the issue is. Diff between newest version and the version that worked for me:
```diff
126c126
<         if self.pointing_device.hid_pending and keyboard._hid_send_enabled:
---
>         if self.pointing_device.hid_pending:
128d127
<             self.pointing_device.hid_pending = False
```

I am running kmk on a pico if that helps.

xs5871, 2022-08-01:
Are you using the `mouse_keys` module in conjunction with another pointing device module?

electricWah, 2022-08-05:
Nope.

xs5871, 2022-08-05:
Hm, interesting.
I'm looking into refactoring the pointing device code, btw. We should handle that by interfacing with a single class, the same way as the keyboard HID does already.

PR #552 (2022-08-05): fix mouse_keys not working at all
resolves #534.
Quick and dirty fix to get mouse keys working at all, at least.
closed 2022-08-05 by kdb424