KMKfw/kmk_firmware #539: KC.LT makes MO layers sticky

slabua, 2022-07-30:
I just noticed that some key combination make momentary layer become irreversibly sticky.
Specifically, I have:

Custom keys:
```
LOWER = KC.MO(1)
RAISE = KC.MO(2)
ADJUST = KC.LT(3, KC.LGUI)
```

Layers:
```
BASE
LOWER
RAISE
ADJUST
[...]
```

If from BASE I press ADJUST + either LOWER or RAISE, that layer activates forever.

Am I missing something?

kdb424, 2022-08-09:
That seems like a release bug. Make sure you are on the latest KMK, and if it's still a problem, please provide any more information you can find. If you turn on debug, does it somehow miss the release, or is it possibly not releasing through a blocker on a layer above? More information is needed.

kdb424, 2022-09-09:
Closing for now as no one seems to have reproduced this error. Will reopen if someone else confirms.

closed 2022-09-09 by kdb424

slabua, 2022-11-13:
Hi, sorry I had totally forgot about this issue since I could not use the keyboard recently.
Today I updated the firmware and the problem seems to have been solved by some commit during these months~
Will update again if anything strange happens. Thank you!