KMKfw/kmk_firmware #559: Global RGB Brightness?

203Null, 2022-08-26:
It seems that "val_limit" is only used in https://github.com/KMKfw/kmk_firmware/blob/91f494846aa53da6124974c10668830e0a82f7ad/kmk/extensions/rgb.py#L467

I think it should be added in (make val_limit replace the upper limit)
https://github.com/KMKfw/kmk_firmware/blob/91f494846aa53da6124974c10668830e0a82f7ad/kmk/extensions/rgb.py#L358

Also, the description of increase_val() is inconsistent too. It says "Increases value by step amount stopping at 100" but it actually clamps to an upper limit of 255.

203Null, 2022-08-26:
I think it's also probably better to clamp them in set_hsv() and set_hsv_fill()

203Null, 2022-08-26:
https://github.com/KMKfw/kmk_firmware/blob/4419f13f1c6488e35c56658f3341760eb660be6a/docs/rgb.md#:~:text=Sets%20the%20saturation%20from%200%2D100 in docs it says that the range of sat is 0~100 

but the default for sat is 255
https://github.com/KMKfw/kmk_firmware/blob/91f494846aa53da6124974c10668830e0a82f7ad/kmk/extensions/rgb.py#L93

xs5871, 2022-09-16:
The global effect of `val_limit` is a good point, and easy to fix as per your suggestion.
The docs are another thing... They have been out of date and sync with the code for quite a while. For some reason I thought some PR or another had addressed that in the past, which obviously isn't the case. I can look into that.

I'm all for refactoring the rgb extension, and we could think about merging some of the key codes for other lighting related extensions (brightness is a good example). That'll be a more in-depth undertaking.

xs5871, 2022-09-21:
closing as resolved by #590 and #591.

closed 2022-09-22 by kdb424