KMKfw/kmk_firmware #564: Comply with Black " vs '

LukeDRussell, 2022-08-31:
In [PR561](https://github.com/KMKfw/kmk_firmware/pull/561) klardotsh indicated he'd be OK with moving KMK to be fully compliant with Black's opinionated use of double quotes (`"` vs. `'`). 

The benefit of this change is:
- contributors can run `black` (deliberately or as part of tests and CI) to have all quotes fixed and standardised. Right now double quotes (`"`) need to be changed manually to single quotes.
- Less divergence from a Python "standard" style. _Yes `black` is highly opinionated_.
- 

This Issue is to track that work.

[Introducing Black ](https://black.readthedocs.io/en/stable/guides/introducing_black_to_your_project.html)will be the primary reference. Particularly the guidance to have everything change in one commit, and introcucing a `ignore-rev` file to make `git blame` a little more usefull.

I will do the work to provide a **single** commit that refactors all of KMK. Though of course I will need QA from the maintainers.

klardotsh, 2022-09-01:
For what it's worth, while I support the move to get rid of a weird vestigial exception case I put in for subjective reasons (back while the debate was still warm over that issue in the community), there should not be a "manually change" step unless something changed on Black's side. I distinctly remember Black's auto-fixer handling this due to the stuff in setup.cfg at one point...

kdb424, 2022-09-01:
At this point, I don't really care. I just want it to be a standard. Pick one, I'll just approve the PR assuming there's no actual code issues.

TLDR: I have no opinion

LukeDRussell, 2022-09-04:
> ...there should not be a "manually change" step unless something changed on Black's side. I distinctly remember Black's auto-fixer handling this due to the stuff in setup.cfg at one point...

I don't understand this point. I've started work in draft [PR571](https://github.com/KMKfw/kmk_firmware/pull/571) though.