KMKfw/kmk_firmware #565: info on powersave

dschil138, 2022-09-01:
Hi, I was wondering if anyone could give more info on the power save feature. The docs explain how to enable it, but not what it does or what the considerations or possible downsides are. It comes with enable/disable keycodes - is it meant to be turned on and off when you start or stop using the board? Or is it something that is meant to be left on all the time?

dschil138, 2022-09-01:
That's weird, the docs mention that as an "extra" feature, but you are saying it's actually the only feature?

I'm using the ItsyBitsy NRF Express, and don't have any LEDs or OLEDs, so is there just no use for me currently?

kdb424, 2022-09-01:
It currently does nothing. I have someone looking into low power standby on NRF chips, but it otherwise does nothing at this very moment.

dschil138, 2022-09-01:
ok thank you for the info!

kdb424, 2022-09-01:
Closing for now as I believe this is resolved to the best it will get done here as changes to this are coming that will make this more clear, and actually useful to most people.

closed 2022-09-01 by kdb424

checksummaster, 2023-02-15:
It seem that value of the power pin on the nano should be inverted. When I use as is, the power on leds turn off when I turn off power saving and vice versa, I invert it in power.py and now it work