KMKfw/kmk_firmware #576: [Docs] Sequence keymap spread causes Python error

daveyholler, 2022-09-07:
While following the docs to create my own sequences, I noticed that the following caused a Python error and the board to crash when loading.
 ```py
keyboard.keymap = [...GET_SPICE,...]
```
However, appending an extra comma to the first set of ellipses (`[...` → `[...,`) worked as expected
```py
keyboard.keymap = [...,GET_SPICE,...]
```

---
Relevant Discord discussion: https://discord.com/channels/704394060390989834/976664447139590174/1017150197501001898

xs5871, 2022-09-07:
Please give a complete report of the issue. Discord / matrix are not "official" records.

klardotsh, 2022-09-07:
Unless something in Python changed without me looking, `...` isn't meaningful in Python anyway, meaning it's a shorthand from the docs used to mean "the rest of your keymap goes here". So `GET_SPICE` is a key that you put in the appropriate place, without the `...`.

Side note: Many KMK users (and particularly, most pre-Boardsource's PEG and Lulu projects) are either not in the Discord, or primarily use Matrix, and further, neither are indexable here on GitHub, so definitely copying the relevant discussion here rather than linking is ideal going forward.

daveyholler, 2022-09-07:
@klardotsh Haha. I'm a total n00b to Python. My Javascript muscles interpreted that as some sort of array spread. I might suggest a minor tweak to the documentation to make it more clear? Just thinking it'd save other fools like me a few minutes of head scratching. 

Even something along these lines would be super helpful:

---
 # Sequences

Sequences are used for sending multiple keystrokes in a single action, and can
be used for things like Unicode characters (even emojis! 🇨🇦), _Lorem ipsum_
generators, triggering side effects (think lighting, speakers,
microcontroller-optimized cryptocurrency miners, whatever). If you are still
unsure of what this is, most other vendors call these "Macros", but can do much
more if you wish.

1. Define your sequences in your main keymap file (_e.g._ `main.py`)
2. Simply reference their names in your `keyboard.keymap` array.

### Example
```py
# Define your custom sequences
SEQUENCE_ONE = simple_key_sequence((KC.LGUI(KC.A), KC.DEL))
SEQUENCE_TWO = simple_key_sequence((KC.LGUI(KC.Z))

# Call them by name in your keymap
keyboard.keymap = [KC.Q, KC.W, <other keycodes>, SEQUENCE_ONE, SEQUENCE_TWO]
```

...

---

The Discord discussion was actually pretty uninteresting, but I was chatting with @kdb424 about it in there.

commit 473d54a8 (2022-09-08): docs(sequences): clarity in keymap arrays to not look like splats (resolves #576)
closed 2022-09-08 by kdb424