KMKfw/kmk_firmware #582: TapHold sudden layer activation

daniel-mayovskiy-frequence, 2022-09-12:
I assume this is a glitch/unintended behavior.

I have been **consistently** experiencing layer hold tap activating regardless of the time held, and delaying the output of the board noticeably.

## Description:
At certain points in time, for unknown reasons, the layer is triggered despite the fact that it is not preferred in the KC.LT() and there is an ample amount of time for the timer to decay.

A [video attached](https://user-images.githubusercontent.com/98058973/189735878-560066f3-2153-4315-9e13-d20c288b22ac.mp4) is of interest, displays the issue at about 42 seconds left on a timer.
Despite never pressing 5, it appeared in the output because numpad-5 is on the key that is under the K keycap on a layer that is triggered by holding spacebar.

The weird part about it is that there is clearly a "thinking" time on the side of the keyboard. While there is an _odd chance_ that it's a user error, and that the timer passed when I was holding that linear switch of the spacebar, the letters from the layer appeared on the screen with _significant_ delay after they were pressed. Which I guess adds complexity to this issue, but should be mentioned nonetheless. Then again, I do not believe I've held the key for long enough for the layer to activate.

## Environment
```
Keyboard: Lulu
OS: Linux 5.15, Ubuntu 20.04LTS
Configuration: Peg
```
I do not believe that the issue is with monkeytype itself, or with my computer, because similar behavior was never experienced in QMK.

Here are the contents of the `main.py` on the right side of the board, where the layout is stored.
The keyboard was configured through Peg, no messing with the `main.py` file happened after the map was uploaded through Peg.

```python
from kb import KMKKeyboard
from kmk.keys import KC
from kmk.modules.layers import Layers
from kmk.modules.modtap import ModTap
from kmk.hid import HIDModes
from kmk.handlers.sequences import send_string
import supervisor
from kmk.extensions.peg_oled_Display import Oled,OledDisplayMode,OledReactionType,OledData
from kmk.extensions.peg_rgb_matrix import Rgb_matrix
from kmk.modules.split import Split, SplitSide, SplitType
keyboard = KMKKeyboard()
modtap = ModTap()
layers_ext = Layers()
keyboard.modules.append(layers_ext)
keyboard.modules.append(modtap)
# oled
oled_ext = Oled(OledData(corner_one={0:OledReactionType.STATIC,1:["layer:"]},corner_two={0:OledReactionType.LAYER,1:["1","2","3","4","5","6","7","8"]},corner_three={0:OledReactionType.LAYER,1:["base","raise","lower","adjust","5","6","7","8"]},corner_four={0:OledReactionType.LAYER,1:["qwerty","nums","shifted","leds","5","6","7","8"]}),toDisplay=OledDisplayMode.TXT,flip=False)
# oled
keyboard.extensions.append(oled_ext)
# ledmap
rgb_ext = Rgb_matrix(ledDisplay=[[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,3],[247,76,0]],split=True,rightSide=True,disable_auto_write=True)
# ledmap
keyboard.extensions.append(rgb_ext)
# TODO Comment one of these on each side
#split_side = SplitSide.LEFT
#split_side = SplitSide.RIGHT
split = Split(data_pin=keyboard.rx, data_pin2=keyboard.tx, uart_flip=False)
keyboard.modules.append(split)
# encodercount
# 2
# encodercount
# keymap
keyboard.keymap = [ [KC.ESCAPE,KC.N1,KC.N2,KC.N3,KC.N4,KC.N5,KC.N6,KC.N7,KC.N8,KC.N9,KC.N0,KC.EQUAL,KC.TAB,KC.Q,KC.W,KC.F,KC.P,KC.B,KC.J,KC.L,KC.U,KC.Y,KC.SCOLON,KC.BSLASH,KC.LCTRL,KC.A,KC.R,KC.S,KC.T,KC.G,KC.M,KC.N,KC.E,KC.I,KC.O,KC.QUOTE,KC.LSHIFT,KC.Z,KC.X,KC.C,KC.D,KC.V,KC.LBRACKET,KC.TG(3),KC.K,KC.H,KC.COMMA,KC.DOT,KC.SLASH,KC.MINUS,KC.RGB_TOG,KC.LGUI,KC.LALT,KC.LT(1, KC.SPACE, prefer_hold=False, tap_interrupted=False, tap_time=180),KC.LT(2, KC.BSPC, prefer_hold=False, tap_interrupted=False, tap_time=170),KC.ENTER,KC.DELETE,KC.SPACE,KC.NO,KC.BSPC,KC.NO,KC.NO,KC.NO,KC.ENTER,KC.DELETE,KC.RGUI,KC.AUDIO_VOL_UP,KC.AUDIO_VOL_DOWN,KC.MEDIA_PREV_TRACK,KC.MEDIA_NEXT_TRACK], 
[KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.PGUP,KC.NUMLOCK,KC.KP_SLASH,KC.KP_ASTERISK,KC.KP_MINUS,KC.TRNS,KC.TRNS,KC.PERCENT,KC.EXCLAIM,KC.AT,KC.DOLLAR,KC.TRNS,KC.PGDOWN,KC.KP_7,KC.KP_8,KC.KP_9,KC.KP_PLUS,KC.TRNS,KC.TRNS,KC.AMPERSAND,KC.COLON,KC.TILDE,KC.CIRCUMFLEX,KC.TRNS,KC.HOME,KC.KP_4,KC.KP_5,KC.KP_6,KC.EQUAL,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.END,KC.KP_1,KC.KP_2,KC.KP_3,KC.KP_ENTER,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.SPACE,KC.TRNS,KC.KP_0,KC.KP_DOT,KC.TRNS,KC.AUDIO_VOL_UP,KC.AUDIO_VOL_DOWN,KC.MEDIA_PREV_TRACK,KC.MEDIA_NEXT_TRACK], 
[KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.LEFT_PAREN,KC.RIGHT_PAREN,KC.LEFT_CURLY_BRACE,KC.RIGHT_CURLY_BRACE,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.HOME,KC.LEFT,KC.DOWN,KC.UP,KC.RIGHT,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.LEFT_CURLY_BRACE,KC.RIGHT_CURLY_BRACE,KC.END,KC.TRNS,KC.LBRACKET,KC.RBRACKET,KC.TRNS,KC.TRNS,KC.LGUI,KC.TRNS,KC.TRNS,KC.TRNS,KC.BSPC,KC.TRNS,KC.TRNS,KC.AUDIO_VOL_UP,KC.AUDIO_VOL_DOWN,KC.MEDIA_PREV_TRACK,KC.MEDIA_NEXT_TRACK], 
[KC.N5,KC.ESCAPE,KC.N1,KC.N2,KC.N3,KC.N4,KC.N6,KC.N7,KC.N8,KC.N9,KC.N0,KC.EQUAL,KC.T,KC.TAB,KC.Q,KC.W,KC.E,KC.R,KC.Y,KC.U,KC.I,KC.O,KC.P,KC.LBRACKET,KC.F,KC.LCTRL,KC.A,KC.S,KC.D,KC.G,KC.H,KC.J,KC.K,KC.L,KC.SCOLON,KC.QUOTE,KC.B,KC.LSHIFT,KC.Z,KC.X,KC.C,KC.V,KC.F3,KC.TG(3),KC.N,KC.M,KC.COMMA,KC.DOT,KC.SLASH,KC.MINUS,KC.GRAVE,KC.LGUI,KC.C,KC.SPACE,KC.BSPC,KC.ENTER,KC.DELETE,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS], 
[KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS], 
[KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS], 
[KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS], 
[KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS,KC.TRNS] ] 
# keymap
if __name__ == '__main__': 
    keyboard.go(hid_type=HIDModes.USB)
```

xs5871, 2022-09-15:
Hm. That'll probably be a pain to figure out.
Could you provide the debug output when that happens? I opened a PR for more usefull debug information for holdtap.

daniel-mayovskiy-frequence, 2022-11-15:
> By the way, does this still happen if you disable the peg extensions?

What are "peg extensions"? The `kmk.extensions` import? Haven't tried it without them. I use the matrix to be able to visually register which layer I was on during the issue, for this issue.

It has been a while since I visited this issue, primarily due to the fact I managed to install QMK on that keyboard while this issue was persistent, since it was essential to my workflow. That being said, I can confirm that the issue is firmware-related, rather than hardware related. This issue does not show up during my time on QMK, despite the fact that I've only increased my typing speed since the time of recording the original video.

I'm not saying that to insult this firmware, only to confirm that it is very unlikely the issue is hardware related.

daniel-mayovskiy-frequence, 2022-11-15:
> Hm. That'll probably be a pain to figure out. Could you provide the debug output when that happens? I opened a PR for more usefull debug information for holdtap.

Could I have a link to that PR, and a link to the way of enabling debug and recording the relevant debug information? Thank you!

xs5871, 2022-11-15:
That PR has been merged a long time ago, and "peg" extensions are all the extensions that start with "peg_".