KMKfw/kmk_firmware #594: Add `Permissive Hold` to HoldTap

ChrisChrisLoLo, 2022-09-18:
Similar to https://github.com/KMKfw/kmk_firmware/issues/593, but tracks `Permissive Hold`.

# Context:
Layouts that heavily depend on HoldTap (such as 40% layouts or layouts that use home row mods) tend to use features such as permissive hold, which works well for faster typists or Home Row mods.
Here is a description of how the behavior works: https://precondition.github.io/home-row-mods#permissive-hold. From my quick experiments, I don't think it's currently possible to recreate this kind of behavior with the current flags available to us.

QMK and ZMK have this kind of behavior as `Permissive Hold` or `Balanced Flavor` respectively. ZMK has a nice diagram of "test cases" that Balanced Flavor has: https://zmk.dev/docs/behaviors/hold-tap

Adding `Permissive Hold` behavior to KMK will further maximize it's feature parity to QMK and ZMK, as well as increase viability for 40% boards, will could be very attractive to a growing subset of the keyboard community.

# Proposed Solution Interface
I would imagine having some sort of boolean `permissive_hold_enabled` argument that could be passed into any HoldTap keys would be an acceptable API, though let me know if you had other thoughts

xs5871, 2022-09-19:
https://github.com/KMKfw/kmk_firmware/blob/master/docs/modtap.md#custom-holdtap-behavior
`tap_interrupted` is what you're looking for.

ChrisChrisLoLo, 2022-09-19:
Thanks! I've tried that flag, though I don't know if it has the behavior as defined in QMK and ZMK. 

When I rapidly press `KC.LT(LOWER, KC.SPACE, tap_interrupted=True)` followed by the top left key (which is `KC.Q` in the default layer and `KC.EXCLAIM` in the LOWER layer) I expect (`!`), what I get instead is `⎵q`, or worse, sometimes `q⎵`. 

If `KC.LT(LOWER, KC.SPACE, tap_interrupted=True)` is key A, and `KC.Q` is key B, then my pressing pattern is 
`Press A, Press B, Release B, Release A` to give me `⎵q` (all in very rapid succession with a timeout of 300, increasing the timeout might make it easy to reproduce)

I haven't pinned down the exact pressing for `q⎵`, but tentatively the pattern seems to be 
`Press A, Press B, Release A, Release B`, where the releases are roughly simultaneous. This pattern leads me to end up with text like this if I type to fast
```
So now I'm going tot he store to pick up eggs
```
This has happened frequently enough that I think there is an edge case here. I have a similar layer in QMK on a different board, and I haven't encountered this kind of problem. I don't think it's a case of me swapping up the key presses when typing fast.

What I'm experiencing seems to contradict the current unit tests, but hopefully there's an explanation between the discrepancy between the two

Here's my debug logs with hold tap debug logging (thanks for adding that in, btw). This is the first edge case, where I expect `!`, but get `⎵q` instead
```
536851431 kmk.kmk_keyboard: MatrixChange(ic=19, pressed=True)
536851436 kmk.kmk_keyboard: KeyResolution(key=Key(code=1018, has_modifiers=None))
536851444 kmk.kmk_keyboard: coordkeys_pressed={19: Key(code=1018, has_modifiers=None)}
536851448 kmk.kmk_keyboard: keys_pressed=set()
536851551 kmk.kmk_keyboard: MatrixChange(ic=0, pressed=True)
536851556 kmk.kmk_keyboard: KeyResolution(key=Key(code=20, has_modifiers=None))
536851563 kmk.kmk_keyboard: coordkeys_pressed={0: None, 19: Key(code=1018, has_modifiers=None)}
536851566 kmk.kmk_keyboard: keys_pressed=set()
536851627 kmk.kmk_keyboard: MatrixChange(ic=0, pressed=False)
536851632 kmk.kmk_keyboard: KeyResolution(key=Key(code=20, has_modifiers=None))
536851636 kmk.modules.holdtap: ht_activate_on_interrupt
536851638 kmk.modules.holdtap: ht_activate_tap
536851659 kmk.kmk_keyboard: coordkeys_pressed={19: Key(code=1018, has_modifiers=None)}
536851663 kmk.kmk_keyboard: keys_pressed={Key(code=44, has_modifiers=None)}
536851667 kmk.kmk_keyboard: MatrixChange(ic=19, pressed=False)
536851672 kmk.kmk_keyboard: KeyResolution(key=Key(code=1018, has_modifiers=None))
536851676 kmk.modules.holdtap: ht_deactivate_on_interrupt
536851678 kmk.modules.holdtap: ht_deactivate_tap
536851686 kmk.kmk_keyboard: coordkeys_pressed={}
536851689 kmk.kmk_keyboard: keys_pressed=set()
536851744 kmk.kmk_keyboard: processing timeouts
```

ChrisChrisLoLo, 2022-09-19:
here are the logs that generate `q⎵`
```
94628 kmk.kmk_keyboard: MatrixChange(ic=19, pressed=True)
94632 kmk.kmk_keyboard: KeyResolution(key=Key(code=1018, has_modifiers=None))
94640 kmk.kmk_keyboard: coordkeys_pressed={19: Key(code=1018, has_modifiers=None)}
94662 kmk.kmk_keyboard: keys_pressed=set()
94686 kmk.kmk_keyboard: MatrixChange(ic=0, pressed=True)
94691 kmk.kmk_keyboard: KeyResolution(key=Key(code=20, has_modifiers=None))
94698 kmk.kmk_keyboard: coordkeys_pressed={0: None, 19: Key(code=1018, has_modifiers=None)}
94701 kmk.kmk_keyboard: keys_pressed=set()
94762 kmk.kmk_keyboard: MatrixChange(ic=19, pressed=False)
94767 kmk.kmk_keyboard: KeyResolution(key=Key(code=1018, has_modifiers=None))
94771 kmk.modules.holdtap: ht_activate_tap
94775 kmk.modules.holdtap: ht_deactivate_tap
94783 kmk.kmk_keyboard: processing timeouts
94793 kmk.kmk_keyboard: coordkeys_pressed={0: Key(code=20, has_modifiers=None)}
94797 kmk.kmk_keyboard: keys_pressed={Key(code=20, has_modifiers=None)}
94801 kmk.kmk_keyboard: MatrixChange(ic=0, pressed=False)
94806 kmk.kmk_keyboard: KeyResolution(key=Key(code=20, has_modifiers=None))
94813 kmk.kmk_keyboard: coordkeys_pressed={}
94817 kmk.kmk_keyboard: keys_pressed=set()
94936 kmk.kmk_keyboard: processing timeouts
```

ChrisChrisLoLo, 2022-09-20:
I've been playing around with my keyboard a bit, and I think I might be off the mark with my diagnosis of what's going on. Removing holdtap doesn't solve the behaviour I'm noticing above. I could be wrong again, but maybe it's possible that KMK scanning times are different than QMK, maybe enough so that what worked for me with QMK might be slightly different with KMK. 

Thanks for you patience with this. If you'd like, I can update the HoldTap documents just to add a mapping of `tap interrupted` to `Permissive Hold`, so it's easier for someone from QMK dipping their toes in to know what to tweak

xs5871, 2022-09-20:
According to your logs, you indeed let go of the holdtap key first and the "interrupting" key second.
You may be just a tiny bit to fast for the scanner. If your using the default scanner, it's rate limited to 20ms scanning intervals. You can try to decrease that; depending on your key switches, that can result in bounces.

ChrisChrisLoLo, 2022-09-20:
Thanks xs5871! It could be that; I can play around a bit with the scanning intervals and go from there

closed 2022-09-20 by ChrisChrisLoLo