KMKfw/kmk_firmware #597: RGB Backlighting crashes the board when toggled

anventia, 2022-09-21:
Trying to run a very simple 2x3 board with 6 RGB LEDs, and when I try to toggle the backlight with `KC.RGB_TOG`, the board crashes.

```py
Traceback (most recent call last):
  File "code.py", line 25, in <module>
  File "kmk/kmk_keyboard.py", line 440, in go
  File "kmk/kmk_keyboard.py", line 494, in _main_loop
  File "kmk/kmk_keyboard.py", line 107, in _handle_matrix_report
  File "kmk/kmk_keyboard.py", line 157, in _on_matrix_changed
  File "kmk/kmk_keyboard.py", line 188, in pre_process_key
  File "kmk/kmk_keyboard.py", line 194, in process_key
  File "kmk/keys.py", line 474, in on_press
  File "kmk/extensions/RGB.py", line 530, in _rgb_tog
  File "kmk/extensions/RGB.py", line 406, in off
  File "kmk/extensions/RGB.py", line 273, in set_hsv_fill
  File "kmk/extensions/RGB.py", line 297, in set_rgb_fill
AttributeError: 'tuple' object has no attribute 'fill'
```

xs5871, 2022-09-21:
That looks familiar. Are you using a recent `adafruit_neopixel` library?
If so, please share the part of your config where you're defining the RGB stuff.

anventia, 2022-09-22:
I'm using the latest version of neopixel.py from Adafruits GitHub.
```py
from kmk.extensions.RGB import RGB

keyboard = KMKKeyboard()

keyboard.col_pins = (board.GP0, board.GP5, board.GP26) 
keyboard.row_pins = (board.GP7, board.GP8)   
keyboard.diode_orientation = DiodeOrientation.COL2ROW

rgb_ext = RGB(pixel_pin=board.GP14, num_pixels=6)
keyboard.extensions.append(rgb_ext)
```

xs5871, 2022-09-22:
I created PR #599 to help debug this. Your configuration looks fine, but I don't see that error with my boards, all with recent KMK and Neopixel libs.

anventia, 2022-10-10:
It appears I accidentally downloaded an old version of the NeoPixel library for some reason. I downloaded the new one from [here](https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/blob/main/neopixel.py) but now I get a different error:
```
Traceback (most recent call last):
  File "code.py", line 25, in <module>
  File "kmk/kmk_keyboard.py", line 440, in go
  File "kmk/kmk_keyboard.py", line 494, in _main_loop
  File "kmk/kmk_keyboard.py", line 107, in _handle_matrix_report
  File "kmk/kmk_keyboard.py", line 157, in _on_matrix_changed
  File "kmk/kmk_keyboard.py", line 188, in pre_process_key
  File "kmk/kmk_keyboard.py", line 194, in process_key
  File "kmk/keys.py", line 474, in on_press
  File "kmk/extensions/RGB.py", line 530, in _rgb_tog
  File "kmk/extensions/RGB.py", line 406, in off
  File "kmk/extensions/RGB.py", line 273, in set_hsv_fill
  File "kmk/extensions/RGB.py", line 297, in set_rgb_fill
TypeError: function takes 2 positional arguments but 3 were given```