KMKfw/kmk_firmware #598: Profile layer keys on spli

kdb424, 2022-09-22:
Some users have comment that layer keys, including MO, feel sluggish compared to QMK. This may be unavoidable as we aren't compiled C, but I'd like to see this profiled and possibly reduced. All complaining users were also split users, so something about splits may also need done. This may involve a protocol change, causing a breaking change in favour of speed.

ChrisChrisLoLo, 2022-10-02:
While I've been having sluggishness problems as well with `KC.LT`, I haven't had any problems with MO personally with my split keyboard using my custom I2C scanner, so I don't know if Python alone is the bottleneck here.

Do you have a list of what the keyboards specifically are, and what communication they're using (Bluetooth or UART)? I'm not super knowledgeable on the matter, though I wonder if there's a chance UART is being bit-banged or the like, or if the underlying UART library is written in Python rather than in C (unlike some of the other circuit python libraries).

As a side note, I don't know if it's necessarily the case here, though something that caught my eye in KMK is that it seems like only one key event gets registered per scanner per scan (IIRC). I could see it being an issue if an MO key is pressed with another key in one scan, since it would take one scan to read the MO key and another scan to pick up the other key. It would be a tall ask to change it so that multiple events got read, and there'd probably need to be some testing to confirm it, though it do think it might be worth looking into since in some scenarios you could end up saving 20ms (or whatever the default scanning period is)

ChrisChrisLoLo, 2022-10-02:
If you want to look at the code for the I2C scanner and my keyboard, you can find it here: https://github.com/ChrisChrisLoLo/beyblock20/blob/master/firmware/beyblock20_controller/i2c_scanner.py.

I don't think my code is particularly optimized/performant, though the two things that might be different is that:
- The I2C library might be more optimized for some reason
- Perhaps it's because I have a dedicated scanner for my other half, and that allows for 2 events to simultaneously occur. It's a shot in the dark though, I'm rusty and new with the scanning code, and I'm not sure how splits work under the hood

kdb424, 2022-10-02:
The limit is there for debounce reasons, and AFIK, all modern firmware have some form of debounce. I will say that at least on my boards that I do notice lag on are rp2040 based, and are blazing fast to loop, so it should not be total loop time at least in the cases I have heard from other users on the same board. It's very likely that timer interruption is not working as intended, or it is, and that's the problem. I hope to see this solved, and it's on my short term list of things to look at, but if someone beats me to it, I'm glad to test your changes. I'm also looking reasons why sometimes the whole loop seems to "lag out" in a way that looks like a GC hang, but doesn't seem to be based on my limited testing.

> I wonder if there's a chance UART is being bit-banged or the like, or if the underlying UART library is written in Python rather than in C (unlike some of the other circuit python libraries).

I don't think this is of much issue. This happens in Cpy land insanely quickly, and is popped onto the stack to queue. I'll look into benchmarking that, but many boards can't do i2c, so I'm trying to go at this in ways that don't involve changing the communication method over splits. The protocol on top could possibly be more optimized, but not by much as it currently only sends 3 bytes, X/Y/Z and it's hard to go more minimal than that.

ChrisChrisLoLo, 2022-10-02:
I agree with a lot of what you're saying, hopefully the profiling reveals where the bottleneck is.

I also agree that the debounce is necessary, though what I wanted to stress is that there may be scenarios where 2 scans are required to process two simultaneous key events, whereas in QMK only 1 scan would have been necessary. I'm also using an rp2040, which I'm super happy about computationally speaking, though the computation speed could be bottlnecked by a potential extra scan. It doesn't seem like it's the issue in this case though :)

kdb424, 2022-10-02:
Would be thrilled if you are willing to look at the matrix scanning, and add that in. I do believe that could help. We do handle the loop at blazing speed honestly, but that is a factor I didn't consider. Tag me in any PR or whatnot if you happen to do any work on that side. I'll likely target LT and the like in the meantime as I think that's the most likely issue for most, but I agree that it can be further sped up.

kdb424, 2022-11-26:
I can't read. It's in the mod-tap docs. Closing.

closed 2022-11-26 by kdb424