KMKfw/kmk_firmware #604: KC[x].after_press_handler() not assigning function

SethSenpai, 2022-09-27:
Hi there,

I'm iterating through the KC.__cache to assign an after_press_handler to every key in the keymap. However it does not seem to add the handler to the function list. When I "hardcode" the assignment it does work however. See the code below;

```
for x in KC.__cache:
    try:
        KC[x].after_press_handler(led_reset_value)
    except:
        print(x + " is a key without after_press_handler function")

#does not work

KC.Q.after_press_handler(led_reset_value)
#does work
```

I am properly calling something wrong but I can't for the life of me figure out what. Any clues on how to implement this?

xs5871, 2022-09-27:
1. Don't do that. `__cache` is a dunder for a reason.
2. You're iterating over keys (not to be confused with `Key`s), not values -- i.e. 'A','B','1','SPACE', etc.
3. If you want to perform an action on _every_ key event, you should use an extionsion or module.

Side note for us maintainers: apparently `KC.__iter__` doesn't work in CP, because of course CP doesn't implement `__iter__` for dictionaries...

SethSenpai, 2022-09-27:
Whoops, not familiar at all with python coding practices, thank you for the information. I'll have a look at what you suggested.

closed 2022-09-27 by SethSenpai