KMKfw/kmk_firmware #626: Auto Update kmkfw.io

hendrix04, 2022-10-24:
We should bring the generation of kmkfw.io into the main repo so that it can be maintained along side the project. This way we don't get public documentation that is 6+ months out of date.

kdb424, 2022-10-25:
That's been discussed with the site owner @daysgobye but for now, I've just started suggesting that users do not use the site. We had automation that apparently didn't work, so it needs resolved, or taken down.

claycooper, 2023-02-12:
I find the website easier to navigate and think it makes the project look more polished. My vote is to keep it going and I volunteer to at least get the website in sync with this repo as a manual effort.

claycooper, 2023-02-17:
@klardotsh @xs5871 @daysgobye I'm newer to the project and understand that there may be some history that I'm not aware of. As I mentioned above, I find that as a new user of KMKfw I find the website docs much easier to use and feel that it gives the project credibility and a professional appearance. Is the main issue that the website currently lags behind the source files in the main repo or are there other concerns?

Thanks.

xs5871, 2023-02-18:
I don't have more information on that, other than what is already stated here.
What I can say is that a shiny website with incorrect information is the antithesis of credibility and professional appearance.

edaubie, 2023-05-28:
Hi there,

I'm new to all this so my apologies if not doing it the right way.

As a new user, I too went straight to the web site documentation. I've spent hours learning KMK from the site and might have lost hours due to incorrect doc (and I said "might", no argument here). I only found out now about the fact the web site is not up to date by reading this post.

Whatever the reason behind this (good or bad, and I agree that it doesn't look professional), wouldn't it be better to put a warning on the front page saying : "The documentation on this website has not been updated, please refer to the github doc"?

claycooper, 2023-05-28:
@edaubie I've been working on ironing out the website staleness. I pushed what's there now and and pushed another update today. I find the website more usable, and likely better for SEO, than the markdown in the firmware repo but I don't have enough privs to fix the current auto-builder. If we're going to put a warning to not use the website then it would be better to shutter the website and I don't want to see that happen. I'm going to keep working on making the site update sustainably from the docs in the firmware repo.

edaubie, 2023-05-30:
I do agree that the web site is needed. It’s a good promotional material.

As I was researching for a prebuilt board to buy, the site helped me figure out if I would be able to program my keyboard and helped me make my decision.

Maybe say something nicer like "KMK gets better regularly, to check the latest version of the documentation, please go to Github ....(url)"