KMKfw/kmk_firmware #633: Layer-tap causing error

sbkeebs, 2022-10-27:
This is on a Boardsource Blok on Microdox v2. When I define a LT key with any layer, any keycode, and any options, holding that key causes this error. 
```
Traceback (most recent call last):
  File "main.py", line 111, in <module>
  File "kmk/kmk_keyboard.py", line 491, in go
  File "kmk/kmk_keyboard.py", line 559, in _main_loop
  File "kmk/kmk_keyboard.py", line 328, in _process_timeouts
  File "kmk/modules/holdtap.py", line 153, in <lambda>
  File "kmk/modules/holdtap.py", line 214, in on_tap_time_expired
  File "kmk/modules/holdtap.py", line 232, in ht_activate_hold
  File "kmk/kmk_keyboard.py", line 261, in resume_process_key
ValueError: object not in sequence

Code done running.
```

The layer-tap i'm using to test this is `LEQL = KC.LT(1, KC.EQUAL, prefer_hold=True, tap_interrupted=False, tap_time=250)`

sbkeebs, 2022-10-30:
trying this with 7.3.3 on a Xiao RP2040, i'm not having this issue, so this might be a problem with Boardsource's fork

prokopiff, 2022-11-01:
Layer-tap has been going through changes a lot during last month or so.
And IMHO it's generally not a good idea to rely on what peg can generate, it's a good starting point, but it's way easier to do complicated things with code.

sbkeebs, 2022-11-01:
This was in plain KMK that i wrote, not using Peg at all, just for the record

xs5871, 2022-11-02:
Can I get some clarification: You're seeing this issue with a Boardsource fork of KMK, but not with upstream, identical configuration?
FYI: that error suggests that the layer module didn't get registered with the keyboard class.

sbkeebs, 2022-11-02:
Clarification: yes, the one that works is on a different controller (a Xiao RP2040), different board but pretty much the same keymap. I think i registered the layer module properly.

My file is here: https://gist.github.com/sbkeebs/68a53513fe13c25a6bf310b53c1ff299 
Layer is appended on line 30. Layer-taps defs start on line 63.

kdb424 suggested that i raise the issue here, but i can definitely bring this up with Boardsource instead.

xs5871, 2022-11-03:
Looks like you're instantiating the layer module twice, and the second time don't add it to the module array. That fits with the your error report. Remove the line `layers_ext = Layers()`.

sbkeebs, 2022-11-03:
oh heck, you're right. i appreciate it.

xs5871, 2022-11-03:
Does that fix it? Can we close the issues?

sbkeebs, 2022-11-04:
it works. please go ahead and close this <3

closed 2022-11-04 by xs5871