KMKfw/kmk_firmware #639: Held keys when keyboard crashes

Max585t, 2022-11-09:
When my keyboard crashes, I notice that certain keys are being reported as held down and that all keyboard inputs (kmk or otherwise) are wrong. When typing, random key-codes will be reported and when using the mouse all input acts as though control is being held.  This does not stop until I restart my pc.

This isn't the biggest deal in the world as I am usually the cause of my keyboard crashing (testing non-kmk code on my keyboard in code.py) but I think it is important to document it. I'm also not convinced this is a kmk issue and not a circuitpython hid or other issue.

prokopiff, 2022-11-10:
I've had keys stuck in down state like this, but it's always fixed by unplugging the keyboard, so it's possible that the problem is in how your OS handles it. I use Linux.

xs5871, 2022-11-11:
The "keys get stuck" is a known and, specifically on a crash, "expected" behavior. If the firmware crashes, it can't release any keys it reported as pressed. The trick to avoid this is not let your keyboard crash. If the crash is due to something in KMK, open a bug report for that. If it's in your own code, you may need some error handling.
The seemingly random things you describe point towards a control key in the set of stuck keys.

PR #716 (2023-03-03): Clear HID reports on crash
resolves #639
closed 2023-03-03 by xs5871