KMKfw/kmk_firmware #640: Bit confused about i2c config

mitswan, 2022-11-12:
So I am trying to set up a board with both an oled and pimoroni trackball and the way they initialise i2c seems incompatible

I cant for example set up the oled with:
```
keyboard.SCL = board.D3
keyboard.SDA = board.D2
```

and then set up the trackball with:
`i2c = io.I2C(scl=board.D3, sda=board.D2)`

Is there a workaround for this?

xs5871, 2022-11-13:
Do you get an error message, or are the peripherals just not working correctly?
The I2C specification supports multiple endpoints, _but_, that's almost never the case for end-user modules. I2C has very tight tolerances on data line impedance; hobbyist hardware modules are designed with the intention to be the only module in that signal chain and do not play well with chaining. It's doable, but likely you have to do some signal analysis and hardware modification.

mitswan, 2022-11-13:
Sorry, i should have popped the error message in the original message:
```
code.py output:
Starting
Traceback (most recent call last):
  File "code.py", line 22, in <module>
ValueError: D3 in use
```

My understanding is that this would be because `busio.I2C(scl=board.SCL, sda=board.SDA)` is being called twice. Ideally I would think that I should be able to pass an `i2c` object into `OLED()` initialiser

xs5871, 2022-11-13:
Ok. Am I correct in assuming you're trying to use the peg oled module? That doesn't support sharing an I2C bus. We (KMK) don't maintain peg (boardsource) code. We host it in our repo for ease of use only.

mitswan, 2022-11-13:
Ah, yes I was trying to use the peg code. Is there an alternative method or
should I look to build my own oled module?

xs5871, 2022-11-14:
There's currently no alternative. It shouldn't be too hard to modify the peg oled code to also accept an initialized I2C device, if you wanted to take that upon yourself.

mitswan, 2022-11-14:
Fantastic,  thanks!

closed 2022-11-29 by mitswan