KMKfw/kmk_firmware #645: How to unpair bluetooth?

lukeallister, 2022-11-17:
Hello kb wizards. 

I have recently built a handwired keyboard using the [Feather nRF52840 Express](https://www.adafruit.com/product/4062)

Reading through [the code](https://github.com/KMKfw/kmk_firmware/blob/master/kmk/kmk_keyboard.py), I see that I can set a secondary HID type:

```
if __name__ == '__main__':
    keyboard.go(hid_type=HIDModes.BLE, secondary_hid_type=HIDModes.USB, ble_name='ergofan')
``` 

And then I can access the HID switch using `KC.HID`.

This works (and should perhaps be added to the bluetooth documentation) but I'm trying to figure out how to _unpair_ the bluetooth so I can connect to a different computer.

In the bluetooth section of the [HID file](https://github.com/KMKfw/kmk_firmware/blob/master/kmk/hid.py) I see 

```
    def clear_bonds(self):
        import _bleio

        _bleio.adapter.erase_bonding()
```

But how to actually do this in the `main.py` file?

I can do it from the circuitpython repl manually using:

```
>>> import _bleio
>>> _bleio.adapter.erase_bonding()
```

Then I can pair it again. 

But how would I get this into a shortcut so I could unpair/reset bluetooth from the keyboard itself?

Any help greatly appreciated.

xs5871, 2022-11-18:
That sounds to me like:
We should add a special function key to disconnect from a BT pairing, like `KC.BLE_DISCONNECT`. Is that an appropriate summary?

lukeallister, 2022-11-19:
Yes, that sounds about right.

greyhatmiddleman, 2022-12-18:
So it seems like there was a possible merge to have multiple bluetooth connection, based on the following:
https://github.com/KMKfw/kmk_firmware/pull/167/files

But the following pull requests/merges dropped those changes.  When I have time I'll fork and see if these changes could be applied with the most recent master.

PR #701 (2023-02-17): Add BLE_DISCONNECT key
resolves #645
closed 2023-02-17 by xs5871