KMKfw/kmk_firmware #646: Rolling from ModTap key to Layer tap key (KC.LT) results in keypresses being emitted out of order

DBendit, 2022-11-21:
# Issue
When rolling from a ModTap key to a Layer tap key, the Layer tap key is emitted before the ModTap key.

# Replication instructions
Using the following config: https://gist.github.com/DBendit/d094873364e37f36d80ccf9f71f2d8ac

Roll from 'l' to 'a'. 'l' is configured as a ModTap key (i.e. KC.MT), and 'a' is configured as a Layer tap key (i.e. KC.LT).

# Expected results
The output should be 'la'.

# Actual results
The output is instead 'al', with the following debug logs: https://gist.github.com/DBendit/f9906968246de3e8a1da4ec59f9d6e67

# Misc
I originally ran into this without `tap_interrupted=True, prefer_hold=False` on the KC.LT keys, but adding those setting didn't help either.

DBendit, 2022-11-21:
Workaround: Rather than using `KC.LT`, use `KC.MT(KC.Key, KC.MO(<layer>))`. This is effectively what the layer module does anyway, but it keeps all of the logic inside of the ModTap module.

xs5871, 2022-11-21:
`tap_interrupted=True` is almost definitely not what you want. Only pass `prefer_hold=False` to both `KC.LT` and `KC.MT`, that should do the trick.

DBendit, 2022-11-21:
@xs5871 You're right - getting rid of the `tap_interrupted=True` params from all of the mod/layer tap keys did get rid of the issue. It still doesn't explain why the behavior seen above happens differently for ModTap-only rolls vs ModTap-to-layer-tap rolls.

xs5871, 2022-11-24:
I don't have an immediate answer to that, and if you're issue is fixed, I'm not going to further investigate it for the time being.