KMKfw/kmk_firmware #650: RFC: Shutdown of #kmkfw:klar.sh Matrix room

klardotsh, 2022-11-24:
Hi everyone!

Let's just jump right in, no big opening blurb from me this time:

- I'm flattered and humbled by how many folks want to use KMK, and want to chat about it
- For the past four years, I've been hosting the synchronous chat systems on my own infrastructure, using Matrix [because free software needs free tools](https://mako.cc/writing/hill-free_tools.html) and I wasn't willing to let the Discord monoculture within the mechanical keyboard scene take another project
- For several reasons, I am intending to gradually exit the Matrix homeserver hosting game:
    - The canonical homeserver implementation, Synapse, is extremely resource intensive
    - The alternatives, such as Dendrite, seem to be in a perpetual state of catching up, and not production-ready
    - I'm less interested in large web-scale federated IM these days, preferring [small dark forests of the internet](https://scribe.citizen4.eu/the-dark-forest-theory-of-the-internet-7dc3e68a7cb1), and thus I am likely switching my personal DMs over to some other non-federated protocol (IRC, Zulip, XMPP without federation, whatever)
    - My plans next summer summarize to meaning my availability (and motivation) to do server maintenance will be limited-if-existent at all, which makes me a horrible single point of failure for an ever-growing online community

In light of all of this, I need to make the unilateral decision to responsibly shut down the `#kmkfw:klar.sh` Matrix channel. What we do from here is where the discussion begins. Here are some options I've thought of. I invite the community to come up with more, and to some degree am offering @xs5871 the final decision making authority here, if they want it.

In relative order of difficulty or disruption:

- Find a new homeserver for the `#kmkfw` Matrix channel, on Matrix. This might mean the flagship instance, or some FOSS-friendly instance, or maybe even an instance hosted by someone in the community. Matrix supports room moves; we could gracefully move all users in the existing room over to the new one and leave a redirect in place for a while at the Matrix level while documentation and so forth gets updated. t2bot.io Discord bridge needs migrated to new room, and moderation permissions set up.
- Migrate fully to the existing Discord guild, free-tools-for-free-software be damned. Remove the t2bot.io Matrix bridge, promote a new moderator in the guild on the Discord side, and life carries on like nothing happened for folks who joined on the Discord side. Matrix users who strongly oppose the privacy or freedom practices of Discord are left out to dry.
- Move official KMK chatter to another IM protocol:
    - IRC (esper.net, oftc.net, libera.chat, etc.)
    - Zulip (disclosure: this is my employer). Can be self-hosted, or use the managed instances at Zulip Cloud, for which sponsorships are available to many FOSS projects (I have nothing to do with that process personally, I just write code here 😄)
    - XMPP (I have no idea how this ecosystem looks these days)
    - Something else?
- Move official KMK chatter to an asynchronous protocol and shut down "official" IM entirely, which makes this issue somewhat of a sibling to #437, for which discussion is yet-ongoing. Unofficial IM rooms could of course still pop up under their own moderation rules, which could potentially even be linked from the README. This is the approach the [Zig programming language](https://ziglang.org) uses: there are **no** official communication platforms other than GitHub over there, but the website links communities which follow a shared code of conduct.

Thanks all for a great four years of KMK so far. I hope whatever we decide here (and in #437) helps set the project up for another four.

klardotsh, 2022-11-24:
I realize I've forgotten to write out a sunset date, in part because I don't yet have one in mind (pending this discussion). If we can tentatively aim for Feb 14, 2023 or sooner, I'd love that (as my volunteering schedule picks up heavily in late-Feb through mid-April, and immediately following that, I tentative set sail north into remote coastal British Columbia for the summer)

Lykos153, 2022-11-24:
Hi :) I'm very new, just chimed in today. Thought I would just be a silent listener for weeks to come, but suddenly a topic pops up I happen to be familiar with :D

This RFC is based on the assumption that Matrix rooms "live" on the home server on which they were created. However, this is not the case. Rooms are completely decentralized and live on all participating servers at the same time. You could shutdown your server today and the room would continue to exist for everybody else.The only ones impacted by the shutdown would be the users of your home server. Well, and of course the room would be left without an admin. So the first two options you mentioned would become just:

* Add a new public address to the Matrix room, promote new moderators in both the Matrix room and the guild on the Discord side, and life carries on like nothing happened for folks who joined either.

The new moderator on Matrix could be you with a new account on a different server. Then nothing would change, except you don't host your own home server anymore.

xs5871, 2022-11-26:
In #437 zulip seems to enjoy popularity. If possible I'd strongly prefer having one single solution, instead of two.

klardotsh, 2022-11-27:
Posting this to both issues (#437 and #650):

[The KMK Zulip instance](https://kmkfw.zulipchat.com) now exists, and allows registration with email, GitHub OAuth, Google OAuth, or GitLab OAuth. Once xs5871 is in the instance, they will be promoted to an equal administrator alongside me.

I've requested FOSS sponsorship to enable things like [public permalinks](https://kmkfw.zulipchat.com/help/public-access-option) which should fully bridge the gap between IM and "forums-style" chat: linking Zulip discussions in GitHub Issues would allow any future readers to catch up on context without signing up for an account on the Zulip instance, for example. I'll report back with the outcome of my sponsorship request (I imagine, given that I now work for Zulip and also given that our Matrix room historically is relatively slow-paced, there should be little drama in that request).

mickeydarrenlau, 2022-11-27:
Why not Gitter instead
It's a centralized platform
And allows any matrix client to join via their matrix server which is hosted by them
@klardotsh

klardotsh, 2022-11-27:
1. What does "a centralized platform" offer us in this circumstance?
2. Matrix compatibility is nice for continuity, admittedly.
3. Matrix does not solve for the inherent weaknesses of traditional room-based IM (Matrix, Slack, Discord, XMPP, IRC, etc.) as discussed in #437. Given an opportunity to merge the two issues into one and solve multiple problems, I'm inclined to weight a solution offering both quite heavily.

klardotsh, 2022-11-27:
Regarding

klardotsh, 2022-11-27:
Regarding [my earlier comment](https://github.com/KMKfw/kmk_firmware/issues/650#issuecomment-1328195786) Zulip Cloud FOSS sponsorship was granted this morning, the linked instance above now has features like public-anonymous permalinking enabled.

mickeydarrenlau, 2022-11-27:
The interface is a bit weird to use

crazyiop, 2022-11-29:
Hi ! I'm voting to leave the discord channel alive.

I personally only casually check in the discord (and tried to help a few times) just out of convenience as I already use it. I'm interested in kmk but not involved and engaged enough to get yet another IM app to follow (I'm not participating in some  community I'm interested in as It seems I would need 10+ different app... So I kind of settled to only the most used one).

I don't know what portion of the people are only on discord. I see 485 (108 online + 377 offline) users in discord right now, but don't know how many are on matrix... (If someone can provide the number, I'm curious now). I think a part of the discord users are also there mostly for the simplicity as the engagement level is very low when you already have the app, which is already widely spread.

klardotsh, 2022-11-30:
I don't have a great way to count Matrix-specific users right now; I'm showing somewhere in the ballpark of 500 total users in the existing chat system, but that counts Matrix native users and Discord-bridged users all the same.

From a quick non-scientific hover-over of random names in the sidebar (where I got that ~500 count from), it appears 50-66% of users in the room may be Matrix-native, and the remainder Discord-bridged.

> the engagement level needed is very low when you already have the app

I have complicated opinions on whether "low engagement level needed" is actually a positive trait - it would likely derail the thread too far for me to get into them, but suffice to say I'm leaning to mild disagreement here. Stopping to think or to get acclimated to a community before talking is not, in my view, a necessarily bad thing, particularly if no information is chat-app-exclusive without public-anonymous links to access it (for a counter-example, I have many bones to pick with the Android ROM scene for largely being Telegram-centric, an app that requires an account to view any information or download any files, and thus without a Telegram account, many Android ROMs are simply undownloadable. KMK will, I hope, never be anywhere near such a condition.)

LukeDRussell, 2022-12-03:
I say go ahead and close both the Discord and Matrix room, leaving just Zulip. I've been vocal against Chat of all kinds though (#437), so you know where I stand.

You could work to get other home servers hosting the KMK channel, which is fine I suppose. Though that would mean we don't get the awesome `topic` feature of Zulip. I've only had a small play with Zulip so far, but it seems like a great improvement to being follow a conversation even in async.

klardotsh, 2022-12-07:
My current leading idea, unless there's significant well-reasoned pushback, is to *only officially endorse the Zulip chat* as a support medium or asynchronous planning platform (aside from GitHub Issues where it's the more appropriate medium). Matrix, Discord, or other platforms are welcome to have their own unofficial KMK communities, under their own moderation standards, with no expectation of maintainers being there (unless they happen to choose to be their own their own terms). I currently do not have plans for setting up a changing of the guard on Matrix or Discord as they currently exist: those rooms are de-facto unmoderated right now anyway, and I have not been active enough in them to keep watch on who should be trusted with moderation going forward. It would be reckless to simply hand "the keys", as it were, to completely random folks, to a room that could be construed as "endorsed" given the channel history, and as such, the likely path is to simply disband the Discord guild, mark the Matrix room as deprecated, and let new communities with active moderation spring up in their place wherever they are (if, of course, they don't move to the Actual Supported And Endorsed Place, https://kmkfw.zulipchat.com)

- @xs5871 thoughts as the only other person with write access to the repo currently?
- @ everyone else, thoughts on what is certainly going to be a controversial issue and, frankly, probably a royal pain in the ass, but a PITA I hope I've explained to a reasonable degree so far?

xgpt, 2022-12-08:
Has anyone recommended the forum features available in github?

klardotsh, 2022-12-09:
Those came up in #437. I personally pushed back on it, but I think the overall vote appeared to be one of indifference, and it's not a hill I'm necessarily willing to die on.

klardotsh, 2022-12-28:
This migration is functionally "done" in the sense that the Matrix and Discord rooms are shut down for all practical/useful purposes, with guiding messages all over the place (topic channel, the last message in the channel, etc.) pointing to the correct place in Zulip chat.

[image]

closed 2022-12-28 by klardotsh