KMKfw/kmk_firmware #651: Support of multi byte keys

tflori, 2022-11-28:
Maybe I'm missing something but I get an error when trying to send the key `0x192`:
>   File "kmk/kmk_keyboard.py", line 491, in go
>   File "kmk/kmk_keyboard.py", line 557, in _main_loop
>   File "kmk/kmk_keyboard.py", line 106, in _send_hid
>   File "kmk/hid.py", line 113, in create_report
>   File "kmk/hid.py", line 183, in add_key
> OverflowError: value must fit in 1 byte(s)

According to evtest on another keyboard the calculator button send `0x192` - that fits to this description:
http://www.freebsddiary.org/APC/usb_hid_usages.php

According to qmk firmware the `KC_CALC` sends a `0xb2` but no matter if I create the key with `Key(code=178)` or `ConsumerKey(code=178)` it does not open a calculator while `0x192` seems to work in windows and linux.

Beside that it seems to be possible to send multi byte keys for other keyboards. Why KMK can't do it? 

I'm not really sure how to change the code to create a PR. When I understand it correct add_key iterates over a string byte for byte and if the byte is `0x00` it replaces the byte with the `key.code` - as my code is above 255 it fails...

My approach would be to just remember the index where to place it and replace it (I'm not coming from python so here is pseudo code):
```
replace <length of key.code> bytes in where_to_place starting from idx with key.code
```

But did I mention that I have no clue what is happening here as that is the first hid for me? it seems that `where_to_place` has a specific length and it could be that there is no more bytes left...

xs5871, 2022-11-28:
Clarification: There's no such thing as multi-byte keys. What you're referring to are consumer control codes.
Consumer Control is its own HID class, with a report different from the keyboard HID, but you've been looking at the code for the keyboard HID.

I noticed that KMK has a faulty implementation of CC when I started with #625. Why is that the case? Probably because until now only single byte codes were ever used and noone noticed. I can look into that once I manage to finishing #625.

tflori, 2022-11-28:
I don't get the `self.report_keys` and how the consumer keys differ.. Glad you can look into it :smile: 

Am I right that a consumer key has 2 bytes and the first is just `0x00` in case of the lower scancodes?

tflori, 2022-12-02:
I'm happy to contribute when I just would know what has to be done :)

do we have to register the HID with 2-byte size and send then every 1-byte code prefixed with `0x00`?

where would we do so? would it not be better two have a different hid class for each device type?

xs5871, 2022-12-03:
Yes, you can help. I opened a draft PR (depends on the HID refactor mentioned above) that you can test.

tflori, 2022-12-03:
yes, it works like a charm :)

tflori, 2022-12-03:
by chance I just saw that the keyboard now reports a key release with every status report after using one of the 2-byte consumer keys:
[image]

closed 2023-02-18 by xs5871

doublep21, 2025-04-18:
> yes, it works like a charm :)

so in the end how you did add the calculator key?

tflori, 2025-04-18:
Simple just create a consumer key:
```
Calc___ = ConsumerKey(code=0x192)
```

doublep21, 2025-04-18:
> Simple just create a consumer key:
> 
> ```
> Calc___ = ConsumerKey(code=0x192)
> ```

thank you

PR #659 (2023-02-18): Fix cc hid report
resolves #651
depends on #625