KMKfw/kmk_firmware #665: Add auto shift on key hold similar to QMK's auto_shift

CupricReki, 2022-12-29:
Add the ability to hold a key for a predetermined time to send the shifted keypress. 

https://docs.qmk.fm/#/feature_auto_shift

The feature could be improved upon by having the ability to control auto shift enabled by layer. For example, disabling autoshift on my gaming layer where I'll need to be holding keys without them being shift modified.

doesntfazer, 2022-12-30:
I'd be interested in making this happen. I'll take a look tomorrow and see what all would be involved. Might be able to get this to work using existing modules. Might need a maintainers input as well.

CupricReki, 2023-01-07:
What did you find? Doesn't it seem like it would be difficult to implement?

doesntfazer, 2023-01-07:
> What did you find? Does it seem like it would be difficult to implement?

It doesn't seem to be too difficult to implement. It's going to take me some time though. remodeling my basement at the moment and my computer stuff is put up right now. All I have is my work computer. 

But also, I just realized when I got the email who this was. Just looked at your username, it took a minute for it to click, It's funny running into you here though after our heated discussion we had on discord lol.

doesntfazer, 2023-01-07:
Either way I'll send you the PR when I'm done with it. D

doesntfazer, 2023-01-07:
So you can pretty much get the same functionality using modtap. These are the setting that I found worked the best to accommodate quick typing.

```python
ASQ = KC.MT(KC.Q, KC.LSFT(KC.Q), prefer_hold=False, tap_interrupted=True, tap_time=180)
ASW = KC.MT(KC.W, KC.LSFT(KC.W), prefer_hold=False, tap_interrupted=True, tap_time=180)
ASE = KC.MT(KC.E, KC.LSFT(KC.E), prefer_hold=False, tap_interrupted=True, tap_time=180)
```
I am going to continue to work on a module that would do this. Although it will probably end up being a submodule to holdtap.

CupricReki, 2023-06-13:
With Autoshift enabled, `KC.LT` will not return to previous layer. 

`Autoshift` should be `AutoShift`

`KC.N0` - `KC.N9` Should autoshift to symbols

Thanks for adding this module! Much appreciated.

CupricReki, 2023-11-30:
> It would be helpful if you could set if autoshift was enabled on symbols and/or numbers, like https://docs.qmk.fm/#/feature_auto_shift?id=auto_shift_symbols-predefined-key-group

https://github.com/KMKfw/kmk_firmware/issues/665#issuecomment-1589050645