KMKfw/kmk_firmware #669: Stringy extension doesn't handle anything but KC.

CupricReki, 2023-01-07:
Looks like the stringy extension needs some love. The extension doesn't handle anything but KC.keys. 

https://github.com/KMKfw/kmk_firmware/blob/3238857e1e715b83edd8b249fc0ef2664b11dca4/kmk/extensions/stringy_keymaps.py#L20

Additionally, it doesn't use the Debug utility so it doesn't honor `StringyKeymaps.debug_enable = True`

xs5871, 2023-01-07:
What else should it handle? If I understand the conversation on zulip, you'd want it handle callable keys, like `KC.MO(1)`? That is almost certainly never going to happen. Callable keys "generated on the fly", i.e. aren't stored in the `KC` dictionary. There's no way to look them up, unless you put them in `KC` manually and reference them by an alias. It's also not reasonable to generate them via a string, because you'd have to write a parser that can distinguish between regular keys and callable ones, and I don't that's worth it. Stringy keymaps are considered more of a toy than core functionality anyway.

Agree with the debug issue though.

CupricReki, 2023-01-07:
The issue is that anything besides `KC.key` gets replaced with `KC.NO`. It makes the extension useless if you plan on using any  "argument" keys.

I can see how writing a parser with recursion would be time consuming. Would it be possible to just ignore anything that's not `KC.key` such as `KC.MO(1)`? I understand that nested keys would have to be non-stringy (`KC.key` instead of the stringy `key`) but it would at least make the extension usable for most applications.

xs5871, 2023-01-07:
Could you please provide a code example for illustration? The extension should only replace string objects in the keymap, not key objects. Mixing strings and regular key objects should already be handled correctly.
Defaulting to `KC.NO` for strings that don't correspond to a `KC.key` is the sensible thing to do, otherwise you'd crash the firmware when pressing that key.

CupricReki, 2023-01-07:
Well I'm a fool. My mistake was using expecting `'MO(1)'`  or `MO(1)` to work. `KC.MO(1)` works fine. Just as expected...

closed 2023-01-07 by CupricReki