KMKfw/kmk_firmware #678: Matrix link in repo description

claycooper, 2023-01-16:
The link to Matrix is still in the repo's description. It should probably be changed to link to Zulip instead.

CupricReki, 2023-01-17:
It's also still in the 'about' section

claycooper, 2023-01-24:
[image]

There's a reference on the right side of the top-level repo view. If you have permissions on the repo there's a gear that shows on the top-right of that panel where you can change it to https://kmkfw.zulipchat.com/

xs5871, 2023-01-24:
I would never have noticed that... thanks.
Technically that should point to a project website, maybe it did point to the rendered docs before they got out of sync. Anyway, I set it to zulip for now.

closed 2023-02-12 by claycooper