KMKfw/kmk_firmware #690: Pi Pico RP2040 - issue with anything kmk.modules.layers/encoder

exiledyorkie, 2023-01-26:
Hi 
I've gotten a Pi Pico board and I've followed numeroius tutorials of getting KMK onto a RP2040 and I've followed them all.
Even though they're practically the same, which is why I can not figure out what is wrong.

I tried to join the Discord but the link was dead and so I thought I'd post here.
Not sure where the issue/fault lies
 
This simple layout works for me
`print("Starting")

import board

from kmk.kmk_keyboard import KMKKeyboard
from kmk.keys import KC
from kmk.scanners import DiodeOrientation

keyboard = KMKKeyboard()

keyboard.col_pins = (board.GP15, board.GP11, board.GP13, board.GP12)    # try D5 on Feather, keeboar
keyboard.row_pins = (board.GP16, board.GP17)    # try D6 on Feather, keeboar
keyboard.diode_orientation = DiodeOrientation.ROW2COL

keyboard.keymap = [
    [KC.ENTER,     KC.LCTL(KC.LALT(KC.S)),      KC.LGUI(KC.D),      KC.LGUI(KC.L),
     KC.LCTL(KC.C),     KC.LCTL(KC.V),      KC.LCTL(KC.LALT(KC.P)),      KC.LCTL(KC.LALT(KC.N)),
    ]
]

if __name__ == '__main__':
    keyboard.go() 

`

But whenever I try and import from kmk.modules.layers etc I get the same error. 
Even any example sketch or code from people I get the same error.
Here is a sketch I was trying to figure out but I get the same error

`print("Starting")

import board

from kmk.kmk_keyboard import KMKKeyboard
from kmk.keys import KC
from kmk.scanners import DiodeOrientation

from kmk.modules.layers import layers

keyboard = KMKKeyboard()

keyboard.col_pins = (board.GP15, board.GP11, board.GP13, board.GP12)    # try D5 on Feather, keeboar
keyboard.row_pins = (board.GP16, board.GP17)    # try D6 on Feather, keeboar
keyboard.diode_orientation = DiodeOrientation.ROW2COL

keyboard.modules.append(layers())

TRANS = KC.TRNS
#layer cange key types

#momentaraly change layer - press and hold
#RAISE = KC.MO(1)

#change between layers
#RAISE = KC.DF(1)
#BASE = KC.DF(0)

#will change layer if held but enter code when tapped
RAISE = KC.LT(1, KC.ENTER)

keyboard.keymap = [
    [#layer 0: BASE 
	 RAISE,     KC.LCTL(KC.LALT(KC.S)),      KC.LGUI(KC.D),      KC.LGUI(KC.L),
     KC.LCTL(KC.C),     KC.LCTL(KC.V),      KC.LCTL(KC.LALT(KC.P)),      KC.LCTL(KC.LALT(KC.N)),
    ],
	[#layer 1: NAV 
	 TRANS,     KC.A,      KC.B,      KC.C,
     KC.E,     KC.F,      KC.G,      KC.H,
    ],
]

if __name__ == '__main__':
    keyboard.go() `

This is the same for an encoder.

I'm running a Pico with the Circuit Python UF2 file and I've downloaed the latest zip file from the KMK Github.
Copied the KMK folder and Boot file and saved my file as code.py/main.py

The error is
[image]
[image]

claycooper, 2023-01-26:
On line 13 you're adding your layers object to the list of modules before you create the object on line 14. Swap lines 13 and 14 and you should be working.

exiledyorkie, 2023-01-26:
> On line 13 you're adding your layers object to the list of modules before you create the object on line 14. Swap lines 13 and 14 and you should be working.

Hello. Thanks for the tip.
I did that and get the following error. I've tried swapping certain lines and get the following;
[image]

claycooper, 2023-01-26:
It's the same issue. You're using the keyboard object on line 15 before creating it on line 17.

exiledyorkie, 2023-01-26:
> It's the same issue. You're using the keyboard object on line 15 before creating it on line 17.

That did the trick, thank you.
Could not work around that one. Just need to fettle the code now, thank you.

I get an error with this and followed suit with moving the lines but get the same issue. Any idea what I'm missing?
[image]

exiledyorkie, 2023-01-26:
> Side note: yes, we moved the community discussions from Matrix/Discord to [Zulip](https://kmkfw.zulipchat.com/). That is mentioned in the `README` -- unfortunately we failed to change that everywhere in the documentation. Not your fault, we're working on it. The Zulip chat may be a better place to ask for help in your case. @claycooper gave some good input already and I'm certain we can get you up and running.

Ah I must have missed that! Will take a look now. Thanks for all the help guys

xs5871, 2023-01-26:
It's `import Layers` and `Layers()`, with a capital L.

exiledyorkie, 2023-01-26:
Thanks again for all the help

xs5871, 2023-01-26:
You're welcome. Would you mind if we close this issue and move further support to zulip?

exiledyorkie, 2023-01-26:
Not a problem. Apologies, I'll go to zulip in future.
I'll play around wih the setup now

closed 2023-01-26 by exiledyorkie