KMKfw/kmk_firmware #703: Bluetooth keys not available

ElVit, 2023-02-15:
Hi,
I am trying to write a firmware for a XIAO BLE based numpad.
You can find my code so far [here](https://github.com/ElVit/kmk-firmware-xiao-ble-numpad/blob/main/code.py).

While implementing the bluetooth functionality I realized that the keys **KC.BT_CLR, KC.BT_PRV** and **KC.BT_NXT** are not available anymore.
Were this keys removed or did I miss something in my code?

Instead I found the button **KC.BLE_REFRESH** in the code. 
I could not find it in the documentation. What is this key doing?

While trying to get some help over discord and matrix I also realized that I can not join.
Are the servers still online? Or did simply the invite code expired?

Thanks

xs5871, 2023-02-16:
kmkfw.io hasn't been updated by the site owner since about a year ago, see #626. We do not recommend using it, please use the original documentation in this repository. You can find the link to our Zulip community chat there.

As for the BT keys: they don't exist, not sure if they ever existed. Noones actively working on BT.
`BLE_REFRESH` stops and restarts advertising (at least that's what the code claims).

ElVit, 2023-02-16:
Thanks for your fast reply.
Maybe it would make sense to give a hint on the homepage that it is outdated.
It was really confusing for me since it is the only documentation I found on the internet for KMK firmware.

And also thanks for the clarification regarding Bluetooth.
I tried KMK mainly because of the bluetooth support. 
But since it is not completed yet I will maybe try ZMK.
I really like the ways to write keyboard firmware in python and hope the BLE support will be supported soon.

closed 2023-02-16 by ElVit

xs5871, 2023-02-16:
> Thanks for your fast reply. Maybe it would make sense to give a hint on the homepage that it is outdated. It was really confusing for me since it is the only documentation I found on the internet for KMK firmware.

It is very confusing, but nothing we can do about. There's a hint on the website, but due to the structure it's not the first thing you see.

> And also thanks for the clarification regarding Bluetooth. I tried KMK mainly because of the bluetooth support. But since it is not completed yet I will maybe try ZMK. I really like the ways to write keyboard firmware in python and hope the BLE support will be supported soon.

BT is working, as far as I know. There's just no active development. You can still contribute or request features.

ElVit, 2023-02-17:
Ok sorry my fault. Yes I can confirm that Bluetooth it's working. And I also saw that you merged the "BLE_DISCONNECT key". Thank you. 

It would be nice to also switch the Bluetooth connections. I will create a feature request for that.