KMKfw/kmk_firmware #708: LayerToggle is incompatible with Chords

Roming22, 2023-02-18:
What I'm trying to achieve: A key outputs a char when pressed, another char when pressed along with another key, and changes layer if held for more than a given delay.

I have reproduced the issue in the following layout: [test.txt](https://github.com/KMKfw/kmk_firmware/files/10775296/test.txt)

In this example the keyboard is a split keyboard, each half being 2x2.
There are 3 layers that are declared statically.
Another 4 layers are automatically generated so that holding a key on one half of the keyboard switches the other half to another layer.

xs5871, 2023-02-19:
You have to use the layer-tap key in the combo:
```python
LTA = KC.LT(3, A)
LTB = KC.LT(4, B)
combos = (Combo((LTA, LTB), KC.I),)
```
... or write a custom combo matcher that ignores hold-tap keys.

Roming22, 2023-02-19:
Thanks for the support, that solved it.

To avoid having to declare a gazillion keys I wrote it as:
```
combos = (Combo((keyboard.keymap[0][key1_index], keyboard.keymap[0][key2_index]), KC.I),)
```

closed 2023-02-19 by Roming22