KMKfw/kmk_firmware #719: Suggestion: Decouple (RGB) Keycodes & Libs from Extensions

moritz-john, 2023-02-25:
Hello,
Recently I created two KMK firmware.
One for the [KLOR](https://github.com/moritz-john/kmk-config-klor/tree/master/firmware) and one for the [Splaytoraid](https://github.com/moritz-john/kmk-config-splaytoraid/tree/master/firmware) keyboard.
 While creating both firmware I faced the same issue:

Let's say I want to ship my keymap with as many features as possible.
E.g.: I'd like to put `KC.RGB_TOG` on a layer by default.

I create my `kb.py` & `main.py` file and host them on GitHub.
Someone downloads both files, puts them on their keyboard and...
_it's not working..?_

Turns out the person did not use LEDs in their keyboard build (or just skimmed over the firmware files) so they decided not to uncommend the LED code in `main.py`:

```py
# from kmk.extensions.RGB import RGB
# rgb = RGB(keyboard.rgb_pixel_pin, num_pixels=22)
# keyboard.extensions.append(rgb)
```
which will output the following error:
`ValueError: Invalid key: RGB_TOG` [via picocom]

Your average user will only notice one thing: **my keyboard is not working.**
They will not be someone who has picocom installed and  be able to troubleshoot this issue.

That's my first reason for suggesting to decouple RGB keycodes from the actual extension.

Make it so that e.g.: `KC.RGB_TOG` does nothing when there is no acutal RGB code running.
Add an error message for e.g.: picocom.
But don't stop the whole keyboard from operating at all. 
That's an absolutely in-transparent design decision for the average end user.

___

Second point:

Let's say this user did some reading and managed to uncomment:
```py
from kmk.extensions.RGB import RGB
rgb = RGB(keyboard.rgb_pixel_pin, num_pixels=22)
keyboard.extensions.append(rgb)
```
well, the keyboard is still not working..?
`ImportError: no module named 'neopixel'` [via picocom]

Turns out the user forgot to add `neopixel.mpy` to their libs folder.

Because as soon as the `from kmk.extensions.RGB import RGB` gets imported, it tries to import [neopixel](https://github.com/KMKfw/kmk_firmware/blob/master/kmk/extensions/rgb.py#L113).
Once again this is completly invisible to your average end user. 
All they know is that their keyboard is not working.

Make it so that `RGB` only _tries_ to import `neopixel`. If it's not found throw an error message for picocom without stopping the whole keyboard from operating. 
___

Thank you for yout time. 
Both my KLOR and Splaytoraid keyboard now ship without RGB keycodes in their KMK keymap by default. I also added something like `splaytoraid_rgb = True/False` in `main.py` to help minimize such problems. Furthermore I ship both firmware with neopixel.mpy inclied. I want to accommodate as many users as possible - also those who are not using RGB.

TL;DR: RGB keycodes, missing RGB extensions or RGB libs should not be able to stop the whole keyboard from operating. This is intransparent behavior for the end user.

xs5871, 2023-02-26:
Summary:
1. Don't crash on invalid key exceptions / default to `KC.NO`. That's something we can do and would also eliminate crashing if there's a typo in the keymap.
2. Don't crash when there are third-party dependencies missing. While on the one hand that crash is totally valid and a very good indicator that somethings awry, I did have the plan to move all hardware initialization for modules/extensions from `__init__` to `during_bootup` for a while. There we can at least catch faulty/missing/optional peripherals and maybe even blacklist or remove them at runtime.

moritz-john, 2023-02-26:
**That's a very good summary, thanks!**

> 1. Don't crash on invalid key exceptions / default to `KC.NO`. That's something we can do and would also eliminate crashing if there's a typo in the keymap.

Yes, I guess the "problem" is not only related to RGB keyodes. Using e.g.: `KC.MPLY` without 
```
from kmk.extensions.media_keys import MediaKeys
keyboard.extensions.append(MediaKeys())
```
also stops the keyboard from working.

___

> 2\. There we can at least catch faulty/missing/optional peripherals and maybe even blacklist or remove them at runtime.

Perfect, I think this would greatly improve the user experience!

manna-harbour, 2023-02-26:
I've also run into this issue, needing to add an option for RGB in Miryoku KMK: https://github.com/manna-harbour/miryoku_kmk#rgb.

A related issue is that the different RGB implementations use different keycodes for the same purpose, e.g. `LED_INC` / `RGB_BRI` / `RGB_VAI`. This doesn't appear to be deliberate (in order to support use of multiple implementations simultaneously) as the same code is re-used in some cases (`RGB_TOG`). Things could be simplified for the user by using only one set of codes, so that the keymap wouldn't need to depend on the implementation.

xs5871, 2023-02-28:
> A related issue is that the different RGB implementations use different keycodes for the same purpose, e.g. `LED_INC` / `RGB_BRI` / `RGB_VAI`. This doesn't appear to be deliberate (in order to [...]

That's a different, arguably not high priority issue. Not as high as other things that need refactoring anyway.

moritz-john, 2023-03-11:
Thank you @xs5871 for implementing those changes!

Would it be possible to apply https://github.com/KMKfw/kmk_firmware/pull/735 to the `peg_rgb_matrix` extension as well? 
It still prefents the keyboard from working if `neopixel.mpy` is missing.

```
File "kmk/extensions/peg_rgb_matrix.py", line 1, in <module>
ImportError: no module named 'neopixel'
```