KMKfw/kmk_firmware #720: Can we have clicker support

Snajj, 2023-02-26:
Hello,
I need a high signal output from pico when any key is pressed. 
I will then feed it to a piezo crystal or buffer it with a mosfet and run a solenoid with it to add some fun to my keyboard. 
I leaned that hijacking the matrix to get this signal directly isn't a trivial matter so I need one of the pins of the pico supplying it.
Is this possible with the current KMK? If it is not can it be added? With KMK having so many features is is odd that something this trivial is missing. Thanks.

xs5871, 2023-02-26:
KMK is built upon Circuitpython. You can use any Circuitpython library you want. For example [digitalio](https://docs.circuitpython.org/en/latest/shared-bindings/digitalio/index.html#), or even
[adiopwm](https://docs.circuitpython.org/en/latest/shared-bindings/audiopwmio/index.html#).

Snajj, 2023-02-26:
> KMK is built upon Circuitpython. You can use any Circuitpython library you want. For example [digitalio](https://docs.circuitpython.org/en/latest/shared-bindings/digitalio/index.html#), or even [adiopwm](https://docs.circuitpython.org/en/latest/shared-bindings/audiopwmio/index.html#).

I see, my Python is very rusty. Which function can I use to get a key press signal. self.get_caps_lock seems to get True when Caps Lock is enabled and false when it is disabled. Is there a similar function that gives true on a key press and false when key is let go.

xs5871, 2023-02-28:
CapsLock (and the other locks) aren't propagated through key presses. Those are host-side settings that are send to the keyboard via USB.
The module class implements the method `process_key`, which is being called for every physical button press/release and one of its parameters is `is_pressed`.