KMKfw/kmk_firmware #721: HoldTap not behaving as expected when the interval between keystroke is short

Roming22, 2023-02-26:
## Configuration:

- Multiple layers
- One key on layer[0] on is set as a HoldTap to LT to another layer
- `prefer_hold = True, tap_interrupted = False` is set on the HT key

A minimal testing configuration is provided.

## Expected behavior:
- When another key is pressed shortly after the HT key has been pressed (and not released), it is processed as key press on the other layer.

## Seen behavior:
On key press that closely follow one another, sometimes it work, sometimes it doesn't.

Logs confirm that the order of the key presses is correct (ic5, then ic14).
Both key presses happens in less than 100ms. The timeout is set to 1000ms.
Both key press are registered before anything is output.

## Additional info:
See attached files:
- Heavily commented configuration: [code.txt](https://github.com/KMKfw/kmk_firmware/files/10834987/code.txt)
- Logs when things go my way: [ok.txt](https://github.com/KMKfw/kmk_firmware/files/10834985/ok.txt)
- Logs when they don't: [fail.txt](https://github.com/KMKfw/kmk_firmware/files/10834984/fail.txt)

manna-harbour, 2023-02-27:
I've also noticed in [Miryoku KMK](https://github.com/manna-harbour/miryoku_kmk) that KMK seems to be slower with layer tap than QMK, ZMK, and KMonad, with what should be the equivalent configuration. I'm also using `prefer_hold=True, tap_interrupted=False`.

[𝑥MK](https://github.com/manna-harbour/xmk) might be useful for investigating this type of timing issue. In particular, [record and playback](https://github.com/manna-harbour/xmk#xmk-usage). It can also be used to compare the behaviour of QMK, ZMK, and KMK with the same input.

xs5871, 2023-03-08:
I'm sorry, but that config is impossible to parse, can't help you with that. If you can bring it into a plain keymap we can look into it.

Roming22, 2023-03-09:
Is that better: [code.txt](https://github.com/KMKfw/kmk_firmware/files/10934495/code.txt)?

With this layout the issue is that 't' followed by 'a' will sometimes give 'ta' and sometimes give 'left', event though 't' is fully released before 'a' is pressed.

xs5871, 2023-03-09:
Add combos to the module array before layers and holdtap, and let me know if that's better.

Roming22, 2023-03-10:
[code.txt](https://github.com/KMKfw/kmk_firmware/files/10945842/code.txt)

That seems to help.

closed 2023-03-13 by Roming22