KMKfw/kmk_firmware #737: Keys are applied the active layer at the time of press, not release

Roming22, 2023-03-11:
**Describe the bug**

- Key1 is configured as `KC.LT(3, KC.T,"prefer_hold"= True)` on layer0.
- Key2 is configured as `KC.MT(KC.H, KC.RSHIFT, tap_time=100)` on layer0, and `KC.MT(KC.DEL, KC.DEL, tap_time=100)` on layer3.
- Key1 is pressed.
- Key2 is pressed.
- Key1 is released less than 100ms after the Key2 press.
- Key2 is released.
- KC.DEL is output.

**To Reproduce**
[xs18.txt](https://github.com/KMKfw/kmk_firmware/files/10949319/xs18.txt)
 
**Expected behavior**
The behavior I would like to see: Since Key1 was released before Key2, the sequence should be treated as a roll with output 'th'.

**Debug output**
```
536868258 kmk.kmk_keyboard: MatrixChange(ic=5, pressed=True)
536868264 kmk.kmk_keyboard: KeyResolution(key=Key(code=1032, has_modifiers=None))
536868286 kmk.kmk_keyboard: coordkeys_pressed={5: None}
536868290 kmk.kmk_keyboard: keys_pressed=set()
536868330 kmk.kmk_keyboard: MatrixChange(ic=13, pressed=True)
536868335 kmk.kmk_keyboard: KeyResolution(key=Key(code=1030, has_modifiers=None))
536868344 kmk.kmk_keyboard: coordkeys_pressed={13: None, 5: None}
536868348 kmk.kmk_keyboard: keys_pressed=set()
536868355 kmk.kmk_keyboard: Error in <RGB object at 0x2000c110>.process_key: 'RGB' object has no attribute 'process_key'
536868363 kmk.kmk_keyboard: Error in <RGB object at 0x2000c110>.process_key: 'RGB' object has no attribute 'process_key'
536868366 kmk.modules.holdtap: ht_activate_on_interrupt
536868368 kmk.modules.holdtap: ht_activate_hold
536868374 kmk.modules.layers: active_layers=[3, 0]
Activate layer: 3
536868385 kmk.kmk_keyboard: MatrixChange(ic=5, pressed=False)
536868390 kmk.kmk_keyboard: KeyResolution(key=Key(code=1032, has_modifiers=None))
536868404 kmk.kmk_keyboard: Error in <RGB object at 0x2000c110>.process_key: 'RGB' object has no attribute 'process_key'
536868408 kmk.modules.holdtap: ht_deactivate_on_interrupt
536868411 kmk.modules.holdtap: ht_deactivate_hold
536868417 kmk.kmk_keyboard: coordkeys_pressed={13: Key(code=1060, has_modifiers=None)}
536868420 kmk.kmk_keyboard: keys_pressed=set()
536868427 kmk.modules.layers: active_layers=[0]
Activate layer: 0
536868437 kmk.kmk_keyboard: MatrixChange(ic=13, pressed=False)
536868442 kmk.kmk_keyboard: KeyResolution(key=Key(code=1060, has_modifiers=None))
536868454 kmk.kmk_keyboard: Error in <RGB object at 0x2000c110>.process_key: 'RGB' object has no attribute 'process_key'
536868458 kmk.modules.holdtap: ht_activate_tap
536868460 kmk.modules.holdtap: ht_deactivate_tap
536868465 kmk.kmk_keyboard: coordkeys_pressed={}
536868469 kmk.kmk_keyboard: keys_pressed=set()
536868475 kmk.kmk_keyboard: Error in <RGB object at 0x2000c110>.process_key: 'RGB' object has no attribute 'process_key'
536868484 kmk.kmk_keyboard: Error in <RGB object at 0x2000c110>.process_key: 'RGB' object has no attribute 'process_key'
536868494 kmk.kmk_keyboard: processing timeouts
536868542 kmk.kmk_keyboard: processing timeouts
536868659 kmk.kmk_keyboard: processing timeouts
```

**Additional context**
There's no real reason for the DEL key to be an MT key, but I though that since the key state is undecided at the time Key1 is released, it might help.

`prefer_hold=True` is required so that key1 pressed, key2 pressed, key2 released, key1 released within 100ms is handled as a key press on layer3.

regicidalplutophage, 2023-03-11:
That is covered by setting ```tap_interrupted=True``` unless I somehow misunderstood the issue.

Roming22, 2023-03-12:
Using the following layout I get these behavior.
[xs18.txt](https://github.com/KMKfw/kmk_firmware/files/10951125/xs18.txt)

`tap_interrupted = False`:
```
192692 kmk.kmk_keyboard: MatrixChange(ic=5, pressed=True)
192698 kmk.kmk_keyboard: KeyResolution(key=Key(code=1032, has_modifiers=None))
192720 kmk.kmk_keyboard: coordkeys_pressed={5: None}
192724 kmk.kmk_keyboard: keys_pressed=set()
192794 kmk.kmk_keyboard: MatrixChange(ic=13, pressed=True)
192799 kmk.kmk_keyboard: KeyResolution(key=Key(code=1030, has_modifiers=None))
192809 kmk.kmk_keyboard: coordkeys_pressed={13: None, 5: None}
192813 kmk.kmk_keyboard: keys_pressed=set()
192820 kmk.kmk_keyboard: Error in <RGB object at 0x2000c2f0>.process_key: 'RGB' object has no attribute 'process_key'
192825 kmk.modules.holdtap: ht_activate_on_interrupt
192828 kmk.modules.holdtap: ht_activate_hold
192834 kmk.kmk_keyboard: Error in <RGB object at 0x2000c2f0>.process_key: 'RGB' object has no attribute 'process_key'
192839 kmk.modules.layers: active_layers=[3, 0]
192844 kmk.kmk_keyboard: Error in <RGB object at 0x2000c2f0>.process_key: 'RGB' object has no attribute 'process_key'
Activate layer: 3
192857 kmk.kmk_keyboard: MatrixChange(ic=5, pressed=False)
192863 kmk.kmk_keyboard: KeyResolution(key=Key(code=1032, has_modifiers=None))
192876 kmk.kmk_keyboard: Error in <RGB object at 0x2000c2f0>.process_key: 'RGB' object has no attribute 'process_key'
192880 kmk.modules.holdtap: ht_deactivate_on_interrupt
192882 kmk.modules.holdtap: ht_deactivate_hold
192889 kmk.kmk_keyboard: coordkeys_pressed={13: Key(code=76, has_modifiers=None)}
192893 kmk.kmk_keyboard: keys_pressed={Key(code=76, has_modifiers=None)}
192900 kmk.kmk_keyboard: Error in <RGB object at 0x2000c2f0>.process_key: 'RGB' object has no attribute 'process_key'
192904 kmk.modules.layers: active_layers=[0]
Activate layer: 0
192913 kmk.kmk_keyboard: processing timeouts
192925 kmk.kmk_keyboard: MatrixChange(ic=13, pressed=False)
192930 kmk.kmk_keyboard: KeyResolution(key=Key(code=76, has_modifiers=None))
192942 kmk.kmk_keyboard: Error in <RGB object at 0x2000c2f0>.process_key: 'RGB' object has no attribute 'process_key'
192948 kmk.kmk_keyboard: processing timeouts
192954 kmk.kmk_keyboard: coordkeys_pressed={}
192957 kmk.kmk_keyboard: keys_pressed=set()
193004 kmk.kmk_keyboard: processing timeouts
193012 kmk.kmk_keyboard: processing timeouts
193025 kmk.kmk_keyboard: processing timeouts
```
Roll outputs `DEL`.

`tap_interrupted = True`:
Roll outputs the desired `th`.
```
536861195 kmk.kmk_keyboard: MatrixChange(ic=5, pressed=True)
536861201 kmk.kmk_keyboard: KeyResolution(key=Key(code=1032, has_modifiers=None))
536861223 kmk.kmk_keyboard: coordkeys_pressed={5: None}
536861227 kmk.kmk_keyboard: keys_pressed=set()
536861281 kmk.kmk_keyboard: MatrixChange(ic=13, pressed=True)
536861287 kmk.kmk_keyboard: KeyResolution(key=Key(code=1030, has_modifiers=None))
536861296 kmk.kmk_keyboard: coordkeys_pressed={13: None, 5: None}
536861300 kmk.kmk_keyboard: keys_pressed=set()
536861307 kmk.kmk_keyboard: Error in <RGB object at 0x2000c270>.process_key: 'RGB' object has no attribute 'process_key'
536861315 kmk.kmk_keyboard: Error in <RGB object at 0x2000c270>.process_key: 'RGB' object has no attribute 'process_key'
536861333 kmk.kmk_keyboard: MatrixChange(ic=5, pressed=False)
536861338 kmk.kmk_keyboard: KeyResolution(key=Key(code=1032, has_modifiers=None))
536861353 kmk.kmk_keyboard: Error in <RGB object at 0x2000c270>.process_key: 'RGB' object has no attribute 'process_key'
536861356 kmk.modules.holdtap: ht_activate_tap
536861359 kmk.modules.holdtap: ht_deactivate_tap
536861366 kmk.kmk_keyboard: coordkeys_pressed={13: Key(code=1030, has_modifiers=None)}
536861370 kmk.kmk_keyboard: keys_pressed=set()
536861376 kmk.kmk_keyboard: Error in <RGB object at 0x2000c270>.process_key: 'RGB' object has no attribute 'process_key'
536861384 kmk.kmk_keyboard: Error in <RGB object at 0x2000c270>.process_key: 'RGB' object has no attribute 'process_key'
536861402 kmk.kmk_keyboard: MatrixChange(ic=13, pressed=False)
536861408 kmk.kmk_keyboard: KeyResolution(key=Key(code=1030, has_modifiers=None))
536861418 kmk.kmk_keyboard: Error in <RGB object at 0x2000c270>.process_key: 'RGB' object has no attribute 'process_key'
536861422 kmk.modules.holdtap: ht_activate_tap
536861425 kmk.modules.holdtap: ht_deactivate_tap
[... 16 lines cut ...]
30863 kmk.kmk_keyboard: Error in <RGB object at 0x2000c270>.process_key: 'RGB' object has no attribute 'process_key'
30871 kmk.kmk_keyboard: Error in <RGB object at 0x2000c270>.process_key: 'RGB' object has no attribute 'process_key'
30917 kmk.kmk_keyboard: MatrixChange(ic=15, pressed=False)
30922 kmk.kmk_keyboard: KeyResolution(key=Key(code=1028, has_modifiers=None))
30936 kmk.kmk_keyboard: Error in <RGB object at 0x2000c270>.process_key: 'RGB' object has no attribute 'process_key'
30939 kmk.modules.holdtap: ht_activate_tap
30942 kmk.modules.holdtap: ht_deactivate_tap
30946 kmk.kmk_keyboard: processing timeouts
30953 kmk.kmk_keyboard: coordkeys_pressed={5: Key(code=1032, has_modifiers=None)}
30957 kmk.kmk_keyboard: keys_pressed=set()
30963 kmk.kmk_keyboard: Error in <RGB object at 0x2000c270>.process_key: 'RGB' object has no attribute 'process_key'
30971 kmk.kmk_keyboard: Error in <RGB object at 0x2000c270>.process_key: 'RGB' object has no attribute 'process_key'
31007 kmk.kmk_keyboard: MatrixChange(ic=5, pressed=False)
31012 kmk.kmk_keyboard: KeyResolution(key=Key(code=1032, has_modifiers=None))
[... cut ...]
31049 kmk.modules.holdtap: ht_deactivate_tap
31052 kmk.kmk_keyboard: processing timeouts
31057 kmk.kmk_keyboard: coordkeys_pressed={}
31060 kmk.kmk_keyboard: keys_pressed=set()
31067 kmk.kmk_keyboard: Error in <RGB object at 0x2000c270>.process_key: 'RGB' object has no attribute 'process_key'
31076 kmk.kmk_keyboard: Error in <RGB object at 0x2000c270>.process_key: 'RGB' object has no attribute 'process_key'
31087 kmk.kmk_keyboard: processing timeouts
```
outputs `th` instead of the desired `DEL`

regicidalplutophage, 2023-03-12:
The second block of debug messages shows 2 sequences of keystrokes:

1.
- Press a key with coordinate 5
- Press a key with coordinate 13
- Release a key with coordinate 5
- Release a key with coordinate 13
That's your basic roll and my expectation is that it would output ```th``` because first key is released after the second key is released.

2.
- Press a key with coordinate 5
- Press a key with coordinate 15
- Release a key with coordinate 15
- Release a key with coordinate 5
Looking at your config that should equate to ```BSPC```, does it not?

Roming22, 2023-03-12:
1. We agree that on a roll, the behavior observed is the right one.
2. We agree that when the keys are released in the reverse order, we expect the output to be from the `system_l` layer, not the `alpha` layer. I'm seeing the later with `th` being the output.

regicidalplutophage, 2023-03-12:
I opened #739 where I reproduce this issue in current version of KMK and describe it more concisely. If you feel like I described everything correctly, you can close this issue.

Roming22, 2023-03-13:
Closing the issue in favor of #739

closed 2023-03-13 by Roming22