KMKfw/kmk_firmware #738: [BUG] OneShot keys don't get released when used in combos

eldritch-elder, 2023-03-12:
**Describe the bug**
Oneshot keys (tested with mods in this case) don't get released in combos properly resulting these to be held indefinetly.

Using the oneshot mods directly on a layer gives the expected result on the same configuration.

**To Reproduce**
With the following snippet, pressing `/` and `'` gives me a glorified caps lock i can't turn off
```python
from kmk.modules.oneshot import OneShot
keyboard.modules.append(OneShot())

from kmk.modules.combos import Combos, Chord, Sequence
combos = Combos()
keyboard.modules.append(combos)

combos.combos = [
    # used key sequence doesn't matter, here is the one i use
    Chord((KC.SLASH, KC.QUOTE), KC.OS(KC.LSFT)),
]
```

**Expected behavior**
Well, i expect them to behave like sticky keys, not really much to describe here for me

**Debug output**
Someone else who tried this send me their debug trail:
```
536824147 kmk.kmk_keyboard: MatrixChange(ic=40, pressed=True)
536824153 kmk.kmk_keyboard: KeyResolution(key=Key(code=56, has_modifiers=None))
536824161 kmk.kmk_keyboard: coordkeys_pressed={40: None}
536824165 kmk.kmk_keyboard: keys_pressed=set()
536824170 kmk.kmk_keyboard: MatrixChange(ic=53, pressed=True)
536824175 kmk.kmk_keyboard: KeyResolution(key=Key(code=52, has_modifiers=None))
536824180 kmk.modules.holdtap: ht_activate_tap
536824187 kmk.kmk_keyboard: coordkeys_pressed={40: None, 53: None}
536824190 kmk.kmk_keyboard: keys_pressed=set()
536824209 kmk.kmk_keyboard: processing timeouts
536824245 kmk.kmk_keyboard: MatrixChange(ic=53, pressed=False)
536824251 kmk.kmk_keyboard: KeyResolution(key=Key(code=52, has_modifiers=None))
536824257 kmk.kmk_keyboard: coordkeys_pressed={40: None}
536824262 kmk.kmk_keyboard: keys_pressed={ModifierKey(code=2, has_modifiers=None)}
536824268 kmk.kmk_keyboard: MatrixChange(ic=40, pressed=False)
536824274 kmk.kmk_keyboard: KeyResolution(key=Key(code=56, has_modifiers=None))
536824281 kmk.kmk_keyboard: coordkeys_pressed={}
536824285 kmk.kmk_keyboard: keys_pressed={ModifierKey(code=2, has_modifiers=None)}
536824291 kmk.kmk_keyboard: ReleaseKeyError(ic=40)
536825181 kmk.kmk_keyboard: processing timeouts
```

**Additional context**
I'm using a splaytoraid rev3 with a Helios by 0xCB but i don't really think that matters a lot given others tested it with the same result

eldritch-elder, 2023-03-12:
Seems like the `ReleaseKeyError` isn't exclusive to OneShot keys, `Chord((KC.N, KC.E), KC.BSPACE)` gives me `ReleaseKeyError(ic=18)`

xs5871, 2023-03-15:
Would you mind taking a look at #746? Keep debug turned off or it'll crash -- there's another PR pending.

moritz-john, 2023-03-16:
> Would you mind taking a look at #746? Keep debug turned off or it'll crash -- there's another PR pending.

(_I took part in the original conversation were the issue was discovered._)
I just tested #746 with
```py
Chord((KC.SLASH, KC.QUOTE), KC.OS(KC.LCTRL)),
```
and
```py
Chord((KC.SLASH, KC.QUOTE), KC.OS(KC.LSFT)),
```
it now works for me.

PR #746 (2023-03-16): Fix oneshot as combo result getting stuck
Looks like we applied a key-release twice.

maybe resolves #738
closed 2023-03-16 by xs5871