KMKfw/kmk_firmware #747: About Bluetooth

Vertex-kb, 2023-03-16:
How to switch between Bluetooth and wired models? Is it possible to use PEG in bluetooth mode.

lukeallister, 2023-03-20:
> How to switch between Bluetooth and wired models

When setting the `keyboard.go()` function you can put secondary hid type like so:

```
if __name__ == '__main__':
    keyboard.go(hid_type=HIDModes.USB, secondary_hid_type=HIDModes.BLE, ble_name='KMK')
```

This has it default to usb, but it can switch to bluetooth. The code for toggling is `KC.HID`

I don't know what PEG is, so somebody else will have to comment on your second question.

0x5e, 2025-06-22:
Hi, I'm using esp32s3, the switching failed:
```
536824415 kmk.keys: TRNS: Key(code=1003, has_modifiers=None)
536824416 kmk.keyboard: coordkeys_pressed={67: Key(code=1001, has_modifiers=None)}
536827360 kmk.keyboard: <Event: key_number 1 pressed>: Key(code=1002, has_modifiers=None)
536827379 kmk.kmk_keyboard: Unexpected error: RuntimeError: pystack exhausted
536827380 kmk.keyboard: cleaning up...
Traceback (most recent call last):
  File "<stdin>", line 47, in <module>
  File "kmk/kmk_keyboard.py", line 478, in go
  File "kmk/kmk_keyboard.py", line 326, in _deinit_hid
TypeError: function takes 1 positional arguments but 0 were given
```

0x5e, 2025-06-23:
Update:
Add `CIRCUITPY_PYSTACK_SIZE = 8192` to `settings.toml` fix this issue.