KMKfw/kmk_firmware #749: Backslash for hash substitution due to US and UK keyboard settings

Optimumfireworks, 2023-03-18:
as I'm working on a Window Pc set to eng,UK the backslash is substituted by the hash when either a single keystroke or when contained within a string

the KC codes however don't give option to use anything other than KC.BSLASH or KC.BSLS which both generate # as does using \ within strings

is there an alternative setup to allow UK selection over US as I assume this will alleviate the issue or an alternative KC. code that will generate \

regicidalplutophage, 2023-03-18:
KMK is built on top of Adafruit keypad library and this library assigns names to scancodes based on how they function in US ANSI keyboard.
In short, KMK calls a certain scancode ```KC.BSLS```, but your PC interprets this scancode as ```#```. What KMK calls it has no bearing on how your PC interprets it.

For keys that are completely missing from US ANSI layout, which that funky key near the the left shift is, you can reference [this doc](https://github.com/KMKfw/kmk_firmware/blob/master/docs/en/international.md).
For the rest you'd have to cross reference images of UK and US keyboard layouts.

xs5871, 2023-03-24:
Re-tagging, because it's not a bug, but a feature request for support for non US-ANSI layouts.