KMKfw/kmk_firmware #776: [BUG] parse error: undefined node label 'pro_micro_i2c' (corne / seeduino xiao ble)

Thempra, 2023-05-01:
**Describe the bug**
Error when compile with Github Actions:

```
Run west build -s zmk/app -b seeeduino_xiao_ble -- -DZMK_CONFIG=${GITHUB_WORKSPACE}/config -DSHIELD="corne_right" 
+ west build -s zmk/app -b seeeduino_xiao_ble -- -DZMK_CONFIG=/__w/zmk-config/zmk-config/config -DSHIELD=corne_right
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /__w/zmk-config/zmk-config/zmk/app
-- Adding /__w/zmk-config/zmk-config/zmk/app/boards/shields/corne
-- ZMK Config directory: /__w/zmk-config/zmk-config/config
-- ZMK Config Kconfig: /__w/zmk-config/zmk-config/config/corne.conf
-- Using keymap file: /__w/zmk-config/zmk-config/config/corne.keymap
-- Using keymap file: /__w/zmk-config/zmk-config/config/corne.keymap
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Cache files will be written to: /__w/zmk-config/zmk-config/zephyr/.cache
-- Zephyr version: 3.2.0 (/__w/zmk-config/zmk-config/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: seeeduino_xiao_ble
-- Shield(s): corne_right
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.15.2 (/opt/zephyr-sdk-0.15.2)
-- Found toolchain: zephyr 0.15.2 (/opt/zephyr-sdk-0.15.2)
-- Found Dtc: /opt/zephyr-sdk-0.15.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /__w/zmk-config/zmk-config/zephyr/boards/arm/seeeduino_xiao_ble/seeeduino_xiao_ble.dts
-- Found devicetree overlay: /__w/zmk-config/zmk-config/zmk/app/boards/seeeduino_xiao_ble.overlay
-- Found devicetree overlay: /__w/zmk-config/zmk-config/zmk/app/boards/shields/corne/corne_right.overlay
-- Found devicetree overlay: /__w/zmk-config/zmk-config/config/corne.keymap
devicetree error: /__w/zmk-config/zmk-config/zmk/app/boards/shields/corne/corne.dtsi:65 (column 1): parse error: undefined node label 'pro_micro_i2c'
-- In: /__w/zmk-config/zmk-config/build/zephyr, command: /usr/bin/python3.8;/__w/zmk-config/zmk-config/zephyr/scripts/dts/gen_defines.py;--dts;/__w/zmk-config/zmk-config/build/zephyr/zephyr.dts.pre;--dtc-flags;'';--bindings-dirs;/__w/zmk-config/zmk-config/zmk/app/drivers/zephyr/dts/bindings;/__w/zmk-config/zmk-config/zmk/app/dts/bindings;/__w/zmk-config/zmk-config/zephyr/dts/bindings;--header-out;/__w/zmk-config/zmk-config/build/zephyr/include/generated/devicetree_generated.h.new;--dts-out;/__w/zmk-config/zmk-config/build/zephyr/zephyr.dts.new;--edt-pickle-out;/__w/zmk-config/zmk-config/build/zephyr/edt.pickle;--vendor-prefixes;/__w/zmk-config/zmk-config/zmk/app/dts/bindings/vendor-prefixes.txt;--vendor-prefixes;/__w/zmk-config/zmk-config/zephyr/dts/bindings/vendor-prefixes.txt
CMake Error at /__w/zmk-config/zmk-config/zephyr/cmake/modules/dts.cmake:230 (message):
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  /__w/zmk-config/zmk-config/zephyr/cmake/modules/zephyr_default.cmake:108 (include)
  /__w/zmk-config/zmk-config/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /__w/zmk-config/zmk-config/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:[9](https://github.com/Thempra/zmk-config/actions/runs/4850343611/jobs/8643178126#step:9:10)2 (include_boilerplate)
  CMakeLists.txt:[15](https://github.com/Thempra/zmk-config/actions/runs/4850343611/jobs/8643178126#step:9:16) (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/__w/zmk-config/zmk-config/build -GNinja -DBOARD=seeeduino_xiao_ble -DZMK_CONFIG=/__w/zmk-config/zmk-config/config -DSHIELD=corne_right -S/__w/zmk-config/zmk-config/zmk/app
Error: Process completed with exit code 1.
```

**To Reproduce**
Keyboard Selection:
    16) Corne
MCU Board Selection:
    21) Seeeduino XIAO BLE

**Expected behavior**
No errors

xs5871, 2023-05-03:
That doesn't look like a bug in KMK. Did you by chance mean to submit it to ZMK?

xs5871, 2023-05-12:
Closing, because apparently not relevant to KMK and no response to clarify.

closed 2023-05-12 by xs5871