KMKfw/kmk_firmware #780: [Help] MIDI encoder

rubanunes, 2023-05-02:
I've searched in a lot of places but couldn't get any answers. Is it possible to make a rotary encoder behave like a MIDI fader? If so does anyone know how? 

Thanks.

xs5871, 2023-05-04:
It should be. There are the encoder module or encoder scanner, whichever you prefer, depending on your use-case or preference to keymap layout. Both do basically the same -- emulating a key-press / key-release, and we do have a midi key module. It's pretty much plug-and-play.
If you want more interactive support I highly recommend checking out the zulip chat. That is the right place to ask. We generally reserve github issues for "low level" firmware stuff, like bug reports and feature requests.

rubanunes, 2023-05-04:
Oh nice thanks for the answer, didn't know about the chat but will definitely check it out.

closed 2023-05-04 by rubanunes