KMKfw/kmk_firmware #814: [Enhancement] add defaults settings to hold tap & other relevant mod keys

Irr0n, 2023-05-30:
https://github.com/KMKfw/kmk_firmware/blob/2d860ebcdb4179d1ec89f498e821cafb4f7b94e7/kmk/modules/holdtap.py#L38

could add a new method to change the defaults so that you don't have to specify the argument on every call
`speedymode` or something

Ch4ni, 2023-09-03:
not sure if this works for you, but here's what I'm using to manage my own defaults:

```
# https://kmkfw.io/docs/holdtap/#custom-holdtap-behavior
# prefer_hold = False       # setting to false means that when typing quickly, the next keypress is interpreted as a tap.
#                           # When intending a hold, this is less likely to happen quickly enough to be interpreted as a tap.
#                           # Fewer false positives than when set to true
# tap_interrupted = True    # interrupt the "hold" after another key is released. For things like Emacs this may provide
#                           # problems, but I'm an nvim user, so it's all good
# tap_time = defaultTime    # use the value of the defaultTime variable for tap_time. This is determined by experimentation 
#                           # to see what works best for the way I type
# repeat = defaultRepeat    # use the value of the defaultRepeat variable for repeat behavior. This is determined by
#                           # experimentation to see what works best for the way I type.
adaHT = lambda kcTap, kcHold : KC.HT(kcTap, kcHold, prefer_hold=False, tap_interrupted=True, tap_time=defaultTapTime, repeat=defaultRepeat)
```

... then I just call `adaHT(KC.<tapKeycode>, KC.<holdKeyCode>)` to use those.

_In theory_ you could _probably_ do something more among the lines of:
```
myHoldTap = lambda kcTap, kcHold, **kwargs : KC.HT(kcTap, kchold, tap_time=<myDefault>, **kwargs)
```
if you wanted to retain the semantics of the original

(disclaimer: I haven't tested the second form of this using `**kwargs` ... I can't recall offhand how that would handle a `kw` that's both manually specified _and_ present in `**kwargs`)

(secondary disclaimer: I'm _relatively_ new to CircuitPython and kmk in general; also not affiliated with the project in any way ... just an excited new user :sweat_smile: )

xs5871, 2023-09-04:
That's the way to do it. Personally, I'd use a named function (if you give it name, a lambda is probably not the right choice).
```python
def speedyHT(*args, **kwargs):
    return KC.HT(*args, prefer_hold=True, **kwargs)
```

Ch4ni, 2023-09-04:
Duly noted, thanks!

I sense an update of my own configs on the way!

closed 2023-09-04 by xs5871