KMKfw/kmk_firmware #818: [Enhancement] Generate mpy artifacts only on changed source

xs5871, 2023-05-30:
**Is your feature request related to a problem? Please describe.**
We started building mpy artifacts to download for every PR merge. Those builds are also triggered for documentation or board code updates.

**Describe the solution you'd like**
Look into using [a changed files GH action](https://github.com/tj-actions/changed-files) (see #800) to generate mpy artifacts only when the python source tree has changed.

rrotter, 2023-08-30:
Looks like this is the solution:

https://stackoverflow.com/questions/70902669/github-action-trigger-a-job-only-when-there-is-a-change-in-particular-director

xs5871, 2023-08-30:
Thanks for the reminder. We actually have a pending PR #857 that does that (well not yet, hence pending).

PR #857 (2023-08-31): feat: Only build mpy files when kmk library files changed
close #818
closed 2023-08-31 by xs5871