KMKfw/kmk_firmware #867: [BUG] Invalid key for Locking Caps Lock, Num Lock, Scroll Lock

syenasweta, 2023-08-24:
1097430 kmk.keys: Invalid key: LOCKING_CAPS
1097437 kmk.keys: Invalid key: LOCKING_NUM
1097444 kmk.keys: Invalid key: LOCKING_SCROLL

JeffreyStocker, 2023-08-25:
Taking a quick glance at the code, I'm seeing part of it commented out with the comment 
"   # FIXME: Investigate whether this key actually works, and
    #        uncomment when/if it does."

https://github.com/KMKfw/kmk_firmware/blob/80e10d30cf147ddc14eda045d34d4253c2deec75/kmk/keys.py#L214
https://github.com/KMKfw/kmk_firmware/blob/80e10d30cf147ddc14eda045d34d4253c2deec75/kmk/keys.py#L416

Unless you need one of those to be locking (which I do not know what locking does off the top of my head), try

- 'CAPS_LOCK', 'CAPSLOCK', 'CLCK', 'CAPS'
- 'SCROLL_LOCK', 'SCROLLLOCK', 'SLCK'
- 'NUM_LOCK', 'NUMLOCK', 'NLCK'

claycooper, 2023-12-29:
Closing as stale.

closed 2023-12-29 by claycooper