KMKfw/kmk_firmware #869: [Enhancement] Send hid signal to switch keyboard layer

hxse, 2023-08-27:
For example, I want to run a script in background to check windows title. When I playing a game, Script auto send a usb hid signal to switch keyboard layer. When I exit the game, Script auto switch the layer back. 
I have implemented this function in qmk, but I don’t know how to implement it also in kmk, this is my repo https://github.com/hxse/py-active-windows-keyboard
How does kmk receive hid signals? Also like `raw_hid_receive` `raw_hid_send` in qmk.

xs5871, 2023-08-28:
KMK doesn't have a `raw_hid` endpoint, and there are no plans to implement one.
However: Circuitpython offers a second serial interface over USB on most MCUs. There's the [serialace](https://github.com/KMKfw/kmk_firmware/blob/master/docs/en/serialace.md) module, which is an example of arbitrary code execution via this serial interface. You can use that as a starting point.

hxse, 2023-09-05:
@xs5871 
I successfully implemented this function.
It can automatically switch different layers in the background from detect different windows title.
It works well.

KMK keyboard example code:
https://github.com/hxse/piantor_kmk_firmware/blob/main/serialace2.py

PC background example code:
https://github.com/hxse/py-active-windows-keyboard/blob/main/send_kmk.py

closed 2023-09-05 by hxse

hxse, 2023-09-15:
@xs5871 
I successfully run Serialace on Raspberry Pi rp2040 keyboard, but I can’t use it with another helios rp2040 https://github.com/KMKfw/kmk_firmware/issues/880
I still think raw_hid is more stable, i hope support it. like (qmk_raw_hid)[https://docs.qmk.fm/#/feature_rawhid]

hxse, 2023-09-21:
```
import usb_cdc
usb_cdc.enable(data=True)
```
add code to boot.py solve my question.