KMKfw/kmk_firmware #880: [BUG] Title

hxse, 2023-09-14:
I use helios rp2040, when i use SerialACE send text, i get out of my expetation.
```
send_kmk: bytearray(b'switch_layer 0\n')
result_kmk: b' | 8.2.6\x1b\\'
```
I want to get success, but get `8.2.6\x1b\\`, It seems to be a runtime version, it is so strange, out of my expetation.
Any suggestion.

xs5871, 2023-09-15:
If you submit a bug, please:
1. Use the bug report template and actually fill it out as intended. There's a reason we're asking for specific information. This bug report doesn't even have a title...
2. Make sure it's actually a bug. Asking for help or support is not a bug. If in doubt, bring up your issue in the [zulip chat](https://kmkfw.zulipchat.com/) **first**.

That being said, looks like you're connected to the wrong serial interface.

hxse, 2023-09-15:
@xs5871 I lost my head

closed 2023-09-15 by hxse

hxse, 2023-09-21:
```
import usb_cdc
usb_cdc.enable(data=True)
```
add code to boot.py solve my question.