KMKfw/kmk_firmware #894: [BUG] KC.GESC produces a runtime error - doesn't appear to be defined.

Bob837217, 2023-09-23:
**Describe the bug**
Can't reference KC.GESC in code.py without runtime error in KMK.

**To Reproduce**

Minimal code to produce this error:
```
import time
from kmk.kmk_keyboard import KMKKeyboard
from kmk.modules.layers import Layers
from kmk.keys import KC
keyboard = KMKKeyboard()
layers = Layers()

keyboard.keymap = [
    # Standard (ISO) Layer
    [
        KC.MUTE,
        KC.GESC
    ],
    # Extra Keys Layer
    [
        KC.B,
    ],
]

while True:
    print("Waiting")
    time.sleep(2)
```

**Expected behavior**
I expect the KC.GESC keycode to be available at runtime and to work on the keyboard keymap.

**Debug output**

Here is the runtime error from REPL:

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 12, in <module>
  File "kmk/keys.py", line 457, in __getattr__
  File "kmk/keys.py", line 475, in __getitem__
  File "kmk/keys.py", line 69, in closure
  File "kmk/keys.py", line 762, in make_key
TypeError: can't convert 'int' object to str implicitly

```

Bob837217, 2023-09-23:
The issue is that I was building my CircuitPython directly from "main" branch, which is all alpha level code and not intended for end users.  I switched to latest non main branch "8.2.x", built from there and this problem does not occur.

main branch version which produced this error: "Adafruit CircuitPython 9.0.0-alpha.1-37-g4563c35908 on 2023-09-12; EK68Custom with rp2040"
8.2.x branch version which resolved this error: "Adafruit CircuitPython 8.2.6-5-g37180b28dd on 2023-09-23; EK68Custom with rp2040"

Closing this issue off.

closed 2023-09-23 by Bob837217

pwarren, 2025-02-13:
This still happens on CircuitPython 9.2.4, at least as released for the adafruit KB2040 at: https://circuitpython.org/board/adafruit_kb2040/

rolling back to 8.2.10 fixes the issue.