KMKfw/kmk_firmware #906: [BUG] KC.MO on RP2040-Plus either crashes or doesn't work

nealie, 2023-10-23:
Calling `KC.MO(1)` on an RP2040-Plus (Waveshare Pi Pico clone) crashes if the documented example is used.

When placed before `keyboard.modules.append(Layers())` the code does not crash, but also does not perform the layer 
shift. When placed afterwards, circuitpython enters safe mode.

My code is as follows:

```
import board
from kmk.kmk_keyboard import KMKKeyboard
from kmk.keys import KC
from kmk.scanners import DiodeOrientation
from kmk.modules.layers import Layers

keyboard = KMKKeyboard()

keyboard.col_pins = (
    board.GP0,  board.GP1,  board.GP2,  board.GP3,  board.GP4,  board.GP5,
    board.GP22, board.GP21, board.GP20, board.GP19, board.GP18, board.GP17
)
keyboard.row_pins = (board.GP6, board.GP7, board.GP8, board.GP9, board.GP10)
keyboard.diode_orientation = DiodeOrientation.COL2ROW

SYMBOL = KC.MO(1)   # <--- Does not crash here, but also doesn't do anything.

keyboard.modules.append(Layers())

#SYMBOL = KC.MO(1)  # <--- Crashes here and enters safe mode.

keyboard.keymap = [
    [ # Layer 0
        KC.ESCAPE, KC.N1,   KC.N2,   KC.N3,   KC.N4,   KC.N5,    KC.N6,    KC.N7,    KC.N8,     KC.N9,   KC.N0,     KC.BSPACE,
        KC.TAB,    KC.Q,    KC.W,    KC.E,    KC.R,    KC.T,     KC.Y,     KC.U,     KC.I,      KC.O,    KC.P,      KC.DELETE,
        KC.LSHIFT, KC.A,    KC.S,    KC.D,    KC.F,    KC.G,     KC.H,     KC.J,     KC.K,      KC.L,    KC.SCOLON, KC.QUOTE,
        SYMBOL,    KC.Z,    KC.X,    KC.C,    KC.V,    KC.B,     KC.N,     KC.M,     KC.COMMA,  KC.DOT,  KC.UP,     KC.ENTER,
        KC.LCTRL,  KC.LGUI, KC.LALT, KC.NO,   KC.NO,   KC.SPACE, KC.SPACE, KC.SLASH, KC.BSLASH, KC.LEFT, KC.DOWN,   KC.RIGHT
    ],
    [ # Layer 1
        KC.TRNS,   KC.EXLM, KC.AT,   KC.HASH, KC.DLR,  KC.PERC,  KC.CIRC,  KC.AMPR,  KC.ASTR,   KC.LPRN, KC.RPRN,   KC.TRNS,
        KC.TRNS,   KC.GRV,  KC.TILD, KC.MINS, KC.UNDS, KC.EQUAL, KC.PLUS,  KC.LCBR,  KC.RCBR,   KC.LBRC, KC.RBRC,   KC.INSERT,
        KC.NO,     KC.F1,   KC.F2,   KC.F3,   KC.F4,   KC.F5,    KC.F6,    KC.NO,    KC.NO,     KC.NO,   KC.COLON,  KC.DQUO,
        KC.NO,     KC.F7,   KC.F8,   KC.F9,   KC.F10,  KC.F11,   KC.F12,   KC.NO,    KC.LABK,   KC.RABK, KC.PGUP,   KC.TRNS,
        KC.RCTRL,  KC.RGUI, KC.RALT, KC.NO,   KC.NO,   KC.TRNS,  KC.TRNS,  KC.QUES,  KC.PIPE,   KC.HOME, KC.PGDOWN, KC.END
    ]
]

if __name__ == '__main__':
    keyboard.go()
```

This issue seems to be the same as this issue on the zulipchat forum: [Crash of rp2040 when I add "FN = KC.MO(1)" line to code.py ](https://kmkfw.zulipchat.com/#narrow/stream/358347-general/topic/Crash.20of.20rp2040.20when.20I.20add.20.22FN.20.3D.20KC.2EMO.281.29.22.20line.20to.20code.2Epy)

xs5871, 2023-10-26:
That is most likely a bug in the Circuitpython build you're using. Certainly not KMK.

claycooper, 2023-12-29:
Per update in Zulip: Unable to reproduce.

closed 2023-12-29 by claycooper

EduardoOliveira, 2024-04-01:
I'm having the same behavior...
@nealie did you manage to solve this?

EduardoOliveira, 2024-04-01:
Just to have it documented somewhere, flashing CircuitPython 8.2.0 solves the issue.