KMKfw/kmk_firmware #907: [BUG] Latest kmk_firmware with latest CircuitPython 9.0.0.alpha.2 on ESP32-S3-DevKitC-1-N8 kinda breaks it

Anutrix, 2023-10-29:
**Describe the bug**
Latest kmk_firmware with latest CircuitPython 9.0.0.alpha on ESP32-S3-DevKitC-1-N8 breaks it.
CIRCUITPY drive and even bootmode Drive both stop working. No matter what reset combination I use.
I hear disconnect/reconnect sounds whenever I try any reset/boot buttons combination.

**To Reproduce**
Steps to reproduce the behavior:
* Flash in UF2 bootloader according to docs.
* Drag and drop CircuitPython 9.0.0.alpha for ESP32-S3-DevKitC-1-N8 to set it up.
* CircuitPython works perfectly.
* On CIRCUITPY drive, add zmk folder to root and write sample program in http://kmkfw.io/docs/Getting_Started to code.py.
* Set keys to board.IO6 and board.IO7.
* Try to save/run it which fails with 'keypad.py doesn't have key_count attribute' kind of error.
* Reset the board after few minutes of trying to make it work.
* CIRCUITPY never comes back. Onwards to panicland.....

**Expected behavior**
No errors or crashes to happen. CIRCUITPY drive shows up again.

**Debug output**
Couldn't note it down as CIRCUITPY wouldn't show and nothing else worked.
I am too scared to try again.

**Additional context**
Wasn't sure which project to report but seeing as CircuitPython worked fine for a while before trying ZMK, I posted it here.

xs5871, 2023-10-30:
Citation from any CP download page:
> Alpha development releases are early releases. They are unfinished, are likely to have bugs, and the features they provide may change.

This is not an issue with KMK. If you don't want bugs, both stable build and (usually) the release candidate are good options.

Anutrix, 2023-10-30:
There were some recent fixes to ESP32-S3 which I wasn't sure if they got applied in Stable branch.
It will try with 8.X stable CircuitPython next.

Anutrix, 2023-11-01:
It is indeed issue with KMK not CircuitPython.

It happens whenever there's working HID related code i.e, KMK and you hard reset.
On next hard reset, both HID logic and CIRCUITPY logic seems to run so CIRCUITPY is not visible.

Steps to reproduce:
1. CIRCUITPY drive is visible. Edit code and add good KMK code with keyboard.go() line.
2. Reconnect the USB cable or press button on the board.
3. See CIRCUITPY go away but not come back.

I see following USBLogView as soon as I plug it in once:
[image]

Seems like it tried to connect as drive but disconnected and turned into HID quickly. Which further disconnects.

xs5871, 2023-11-05:
Did you confirm that this also happens with the stable release?

adam-huganir, 2023-11-10:
I am having the same issue with the same board on 8.2.6 stable, i thought it was hardware and just used a different board but since I saw this ticket I thought I'd jump in and confirm

xs5871, 2023-11-10:
Looks like known issue: https://github.com/adafruit/circuitpython/issues/7664

adam-huganir, 2023-11-10:
Awesome, i'll give that a try and get that lil macro pad up and running

> Looks like known issue: [adafruit/circuitpython#7664](https://github.com/adafruit/circuitpython/issues/7664)

claycooper, 2023-12-29:
Closing as stale.