KMKfw/kmk_firmware #931: [BUG] Documentation out of date with current repo?

xgpt, 2024-01-19:
**Describe the bug**
It looks like the current kmkfw.io docs point to a version where there was no internationalization and all of the files were at the root of /docs/PAGE.md instead of /docs/en/PAGE.md etc.

How do we get the online documentation up to date with current files?

claycooper, 2024-01-24:
The docs are being refactored to have a more supportable build process. I'll leave this issue open until we get a somewhat recent version of the docs available on the website.

hendrix04, 2024-07-11:
This and https://github.com/KMKfw/kmk_firmware/issues/626 are duplicates of each other.

claycooper, 2024-07-11:
The documentation is currently up to date, which is the topic of this bug, so I'll close this out. #626 is about automating, which is still in progress, and I'll leave it open.

closed 2024-07-11 by claycooper