KMKfw/kmk_firmware #939: [BUG] Combos with home row modifiers like KC.HT(KC.F, KC.LSFT)

juru1234, 2024-02-23:
**Describe the bug**
Hi, I want to use combos along with home row modifiers.
When I long press F, it should be LSFT.
When I long press D, it should be LCTRL.
When I press F and D as a coombo, it should be ESC.

I tried something like this but it does not seem to work:
`Chord((KC.HT(KC.D, KC.LCTRL), KC.HT(KC.F, KC.LSFT)), KC.ESC)`

Is this feature not implemented in KMK?

xs5871, 2024-02-23:
You have to alias the hold-tap keys and use that alias in the combo and keymap.
Argumented keys are dynamically created; even though two argumented keys (hold-tap in this case) may have the same arguments, they are different objects.

juru1234, 2024-02-23:
Yeah, it works! Thank you so much!

closed 2024-02-23 by juru1234