KMKfw/kmk_firmware #947: [BUG] circuitpython 9.0.0

gsteen, 2024-03-20:
I installed circuitpython 9.0.0 on pi pico.
Followed the Getting_Started.md and got a working 'A'.
Added Layer module -> Hard crash. I cannot debug it.
Installed circuitpython 8.2.10 and it works like expected.

kolya329, 2024-03-22:
Wait... So I was having this crash for like 4 days (nuked pi pico W, because I'm stupid) and that's why?! Why haven't I tried it yet?..
But yes, seems like RP2040 is the problem here. Or maybe we will see this problem later.

For debug purposes, that's what Thonny gives me when layers = Layers() is before KC.LT etc.:

PROBLEM IN THONNY'S BACK-END: Exception while handling 'Run' (ConnectionError: EOF).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your CircuitPython device and try again.

Unable to connect to COM5: port not found
Process ended with exit code 1.

xs5871, 2024-03-27:
I cannot reproduce crashes with CP 9.0.0.
Have you tried a more recent build?

gsteen, 2024-03-27:
I used this Version:
https://adafruit-circuit-python.s3.amazonaws.com/bin/raspberry_pi_pico/de_DE/adafruit-circuitpython-raspberry_pi_pico-de_DE-9.0.0.uf2

Some hours ago CP 9.0.1 was released. I will try it.

lukeallister, 2024-04-04:
Me too on the [YD_RP2040](https://circuitpython.org/board/vcc_gnd_yd_rp2040)

I downgraded to 8.2.10 and it works.

iovxw, 2024-04-06:
`Adafruit CircuitPython 9.0.3 on 2024-04-04; VCC-GND Studio YD RP2040 with rp2040`

kmk version: 46ab5cbc26ee8a63be5217005abc6221d899ad5b

reproduced in REPL:
```python
from kmk.modules.layers import Layers
from kmk.keys import KC
Layers()
KC.TG(0)
```
```
>>> from kmk.keys import KC
>>> from kmk.modules.layers import Layers
>>> Layers()
113470 kmk.keys: Invalid key: HT
113474 kmk.keys: NO: Key(code=1000, has_modifiers=None)
<Layers object at 0x2000df60>
>>> KC.TG(0)

[19:52:14.480] Disconnected
[19:52:15.482] Connected
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```

Looks like a bug in CircuitPython, and I found a fix that works on my board:

```diff
diff --git a/kmk/keys.py b/kmk/keys.py
index 364eccf..aed0244 100644
--- a/kmk/keys.py
+++ b/kmk/keys.py
@@ -807,6 +807,8 @@ def make_argumented_key(
         if meta:
             key = Key(
                 NEXT_AVAILABLE_KEY, meta=meta, *constructor_args, **constructor_kwargs
+            ) if constructor_args else Key(
+                NEXT_AVAILABLE_KEY, meta=meta, **constructor_kwargs
             )
 
             NEXT_AVAILABLE_KEY += 1
```

xs5871, 2024-04-07:
> Looks like a bug in CircuitPython, and I found a fix that works on my board:

Not a bug in CP, that syntax is incorrect. Fixed by #958.
I already found that in my local (heavily modified) copy of KMK and didn't check if that issue existed upstream.
The hard memory faults are not explained by this though.

gsteen, 2024-04-20:
With kmk 5525233 and CP to 9.0.4 it is working :-)

closed 2024-04-20 by gsteen