KMKfw/kmk_firmware #961: Key tap not working as expected

rishabhrkaushik, 2024-04-13:
**Describe the bug**
Trying out an example to send particular keycodes based on conditions in loop. Key is being sent once but in the loop, the keys are not being sent.

**To Reproduce**
Board used is ESP32-S3-N16R8. 
Circuit Python version 9.0.3

```python
import board
import time
from kmk.kmk_keyboard import KMKKeyboard
from kmk.keys import KC

keyboard = KMKKeyboard()

keyboard.debug_enabled = True

# Keycodes
B = KC.B
# C = KC.C

counter = 0

# Main loop function to send keycodes and wait
def main_loop():
    global counter
    while True:
        # Increment the counter each cycle
        counter += 1
        print(f"Sending 'B', Count: {counter}")

        # keyboard.tap_key(B)
        keyboard.keys_pressed.add(KC.B)
        keyboard.process_key(KC.B, True)
        time.sleep(0.1)
        keyboard.keys_pressed.discard(KC.B)
        keyboard.process_key(KC.B, False)
        # keyboard.tap_key(C)
        
        # Wait for 10 seconds
        time.sleep(10)

if __name__ == '__main__':
    main_loop()

```
with the output
```txt
code.py output:
50912 kmk.keys: B: Key(code=5, has_modifiers=None)
Sending 'B', Count: 1
Sending 'B', Count: 2
```

The following getting started code does work
```python
import board

from kmk.kmk_keyboard import KMKKeyboard
from kmk.keys import KC
from kmk.scanners import DiodeOrientation

keyboard = KMKKeyboard()
keyboard.debug_enabled = True

keyboard.col_pins = (board.GPIO4,)
keyboard.row_pins = (board.GPIO5,)
keyboard.diode_orientation = DiodeOrientation.COL2ROW

keyboard.keymap = [
    [KC.A,]
]

if __name__ == '__main__':
    keyboard.go()
```

with the output
```txt
code.py output:
388406 kmk.keys: A: Key(code=4, has_modifiers=None)
388407 kmk.keyboard: Initialising KMKKeyboard
388407 kmk.keyboard: unicode_mode=0
388408 kmk.hid: use 6KRO
388419 kmk.hid: use no pan
388420 kmk.keyboard: hid=USBHID(REPORT_BYTES=9)
388422 kmk.keyboard: matrix=['MatrixScanner']
388423 kmk.keyboard: modules=[]
388423 kmk.keyboard: extensions=[]
388434 kmk.keyboard: mem_info used:50272 free:8204128
398384 kmk.keys: TRNS: Key(code=1000, has_modifiers=None)
398385 kmk.keyboard: <Event: key_number 0 pressed>: Key(code=4, has_modifiers=None)
398386 kmk.keyboard: coordkeys_pressed={0: Key(code=4, has_modifiers=None)}
398387 kmk.keyboard: keys_pressed={Key(code=4, has_modifiers=None)}
398519 kmk.keyboard: <Event: key_number 0 released>: Key(code=4, has_modifiers=None)
398521 kmk.keyboard: coordkeys_pressed={}
```

**Expected behavior**
A clear and concise description of what you expected to happen.
Key B is sent every 10 seconds

xs5871, 2024-04-17:
That's not a bug. If you're not running the main loop with `keyboard.go()`, of course nothing is happening.

xs5871, 2024-05-26:
closing because: not an issue

closed 2024-05-26 by xs5871