KMKfw/kmk_firmware #968: Combos not working on keys that include holdtap or tapdance

KastenTaco, 2024-05-17:
**Describe the bug**
I have include multiple combos to my layout, all that rely on keys that have no tap dance or hold tap modifications, work fine. However, all combos utilizing keys that include such mods, do not trigger. Mods and keymap are included:

**To Reproduce**
```
combos.combos = [
    #Select all
    Chord((KC.A, KC.LALT), KC.LCTL(KC.A)),
    #Apos
    Chord((KC.L, KC.SCOLON), KC.QUOT),
    #CapsWord
    Chord((KC.F, KC.J), KC.CW),
]

# Cleaner key names
_______ = KC.TRNS
XXXXXXX = KC.NO

keyboard.keymap = [
    #[0] QWERT BASE Layer 
    [KC.Q,                               KC.W,                               KC.E,                                 KC.R,                                    KC.T,                             KC.Y,                                       KC.U,                                    KC.I,                                    KC.O,                               KC.P,
     KC.A,                               KC.TD(KC.HT(KC.S, KC.LALT), KC.S),  KC.TD(KC.HT(KC.D, KC.LCTRL), KC.D),   KC.TD(KC.HT(KC.F, KC.LSFT), KC.F),       KC.G,                             KC.H,                                       KC.TD(KC.HT(KC.J, KC.RSFT), KC.J),       KC.TD(KC.HT(KC.K, KC.RCTRL), KC.K),      KC.TD(KC.HT(KC.L, KC.RALT), KC.L),  KC.SCOLON,
     KC.TD(KC.HT(KC.Z, KC.LGUI), KC.Z),  KC.X,                               KC.C,                                 KC.V,                                    KC.B,                             KC.N,                                       KC.M,                                    KC.COMMA,                                KC.DOT,                             KC.TD(KC.HT(KC.SLASH, KC.RGUI), KC.SLASH),
     XXXXXXX,                            XXXXXXX,                            KC.HT(KC.ESC, KC.MO(3)),              KC.TD(KC.HT(KC.ENT, KC.MO(4)), KC.ENT),  KC.HT(KC.TAB, KC.MO(5)),          KC.TD(KC.HT(KC.BSPC, KC.MO(6)), KC.BSPC),   KC.TD(KC.HT(KC.SPC, KC.MO(7)), KC.SPC),  KC.TD(KC.HT(KC.DEL, KC.MO(8)), KC.DEL),  XXXXXXX,                            XXXXXXX,
    ],
```

**Expected behavior**
I was expecting I could use the default tap behavior of each key and link it to another to trigger a combo.
Example: A + S = Select All
S has a tap dance and hold tap modifier attached does not trigger a combo.

**Debug output**
Combo that does _not_ trigger:
```
393750 kmk.keyboard: <Event: key_number 5 pressed>: Key(code=4, has_modifiers=None)
393757 kmk.keyboard: coordkeys_pressed={5: None}
393766 kmk.keyboard: <Event: key_number 6 pressed>: Key(code=1023, has_modifiers=None)
393771 kmk.keyboard: coordkeys_pressed={6: None, 5: None}
393776 kmk.keyboard: coordkeys_pressed={6: None, 5: Key(code=4, has_modifiers=None)}
393779 kmk.keyboard: keys_pressed={Key(code=4, has_modifiers=None)}
393785 kmk.keyboard: coordkeys_pressed={6: Key(code=1023, has_modifiers=None), 5: Key(code=4, has_modifiers=None)}
393808 kmk.keyboard: <Event: key_number 5 released>: Key(code=4, has_modifiers=None)
393814 kmk.keyboard: coordkeys_pressed={6: Key(code=1023, has_modifiers=None)}
393827 kmk.keyboard: <Event: key_number 6 released>: Key(code=1023, has_modifiers=None)
393832 kmk.keyboard: coordkeys_pressed={}
393966 kmk.modules.holdtap: ht_activate_tap
393971 kmk.keyboard: keys_pressed={Key(code=22, has_modifiers=None)}
```
Combo that does trigger:
```
485567 kmk.keyboard: <Event: key_number 13 pressed>: Key(code=25, has_modifiers=None)
485575 kmk.keyboard: coordkeys_pressed={13: None}
485584 kmk.keyboard: <Event: key_number 14 pressed>: Key(code=5, has_modifiers=None)
485589 kmk.modules.combos: activateChord([25, 5])
485593 kmk.keyboard: coordkeys_pressed={14: None, 13: None}
485595 kmk.keyboard: keys_pressed={Key(code=25, has_modifiers={1})}
485645 kmk.keyboard: <Event: key_number 13 released>: Key(code=25, has_modifiers=None)
485648 kmk.modules.combos: deactivateChord([25, 5])
485652 kmk.keyboard: coordkeys_pressed={14: None}
485663 kmk.keyboard: <Event: key_number 14 released>: Key(code=5, has_modifiers=None)
485669 kmk.keyboard: coordkeys_pressed={}
485672 kmk.keyboard: release w/o press:14
485673 kmk.keyboard: coordkeys_pressed={}
```

**Additional context**
From the debugging output, it seems the combo does not trigger.

KastenTaco, 2024-05-18:
I went and redifined home row mods as individual keys. It now works. 

```
#Homerow Mods
# Left side
F_SHIFT = KC.HT(KC.F, KC.LSFT, prefer_hold=False, tap_time=300)
D_CTRL = KC.HT(KC.D, KC.LCTRL, prefer_hold=False, tap_time=300)
S_ALT = KC.HT(KC.S, KC.LALT, prefer_hold=False, tap_time=300)
Z_GUI = KC.HT(KC.A, KC.LWIN, prefer_hold=False, tap_time=300)

# Right side
J_SHIFT = KC.HT(KC.J, KC.LSFT, prefer_hold=False, tap_time=300)
K_CTRL = KC.HT(KC.K, KC.LCTRL, prefer_hold=False, tap_time=300)
L_ALT = KC.HT(KC.L, KC.LALT, prefer_hold=False, tap_time=300)
SLASH_GUI = KC.HT(KC.SLSH, KC.LWIN, prefer_hold=False, tap_time=300)

combos.combos = [
    #Select all
    Chord((KC.A, S_ALT), KC.LCTL(KC.A)),
    #Undo
    Chord((KC.W, KC.E), KC.LCTL(KC.Z)),
    #Redo
    Chord((KC.E, KC.R), KC.LCTL(KC.Y)),
    #Cut
    Chord((KC.X, KC.C), KC.LCTL(KC.X)),
    #Copy
    Chord((KC.C, KC.V), KC.LCTL(KC.C)),
    #Paste
    Chord((KC.V, KC.B), KC.LCTL(KC.V)),
    #Refresh
    Chord((KC.Y, KC.U), KC.F5),
    #Apos
    Chord((L_ALT, KC.SCOLON), KC.QUOT),
    #CapsWord
    Chord((F_SHIFT, J_SHIFT), KC.CW),
]

# Cleaner key names
_______ = KC.TRNS
XXXXXXX = KC.NO

keyboard.keymap = [
    #[0] QWERT BASE Layer 
    [KC.Q,     KC.W,     KC.E,                     KC.R,                                    KC.T,                             KC.Y,                                       KC.U,                                    KC.I,                                    KC.O,                               KC.P,
     KC.A,     S_ALT,    D_CTRL,                   F_SHIFT,                                 KC.G,                             KC.H,                                       J_SHIFT,                                 K_CTRL,                                  L_ALT,                              KC.SCOLON,
     Z_GUI,    KC.X,     KC.C,                     KC.V,                                    KC.B,                             KC.N,                                       KC.M,                                    KC.COMMA,                                KC.DOT,                             SLASH_GUI,
     XXXXXXX,  XXXXXXX,  KC.HT(KC.ESC, KC.MO(3)),  KC.TD(KC.HT(KC.ENT, KC.MO(4)), KC.ENT),  KC.HT(KC.TAB, KC.MO(5)),          KC.TD(KC.HT(KC.BSPC, KC.MO(6)), KC.BSPC),   KC.TD(KC.HT(KC.SPC, KC.MO(7)), KC.SPC),  KC.TD(KC.HT(KC.DEL, KC.MO(8)), KC.DEL),  XXXXXXX,                            XXXXXXX,
    ],
```

xs5871, 2024-05-18:
That is the intended solution. Not a bug, just how KMKs key processes pipeline works.

closed 2024-05-18 by xs5871

Boshimanu, 2024-07-07:
Thank You. This "Non Bug", solved my situation. I Was having the same issue. 🙏

halfdane, 2024-07-18:
Not sure what I'm doing wrong, or how extracting a key that's used nowhere but in the keymap itself changes the behaviour of CapsWord, but here's another minimal example of the problem:

```
R_ALT=KC.HT(KC.R, KC.LALT, prefer_hold=False, tap_interrupted=True, tap_time=150)

keyboard.keymap = [ [
   KC.TD(KC.CW, KC.CAPS, tap_time=200), KC.F, R_ALT,  
] ]
```

Tapping all keys from left to right exactly once will give the following output:
`Fr`
Please note that the second character isn't capitalized, although CapsWord is active.

(Tapping the first key twice will result in the expected output: `FR`)