KMKfw/kmk_firmware #1010: [BUG] KeyError when using TapDance

Crypto-Darth, 2024-07-26:
Hey there,
So I've been making my custom macropad using Raspberry Pi pico (RP2040).I am fairly new to these stuff so forgive me if this was my issue.

Upon running TapDance Function as it is in the [tapdance](https://github.com/KMKfw/kmk_firmware/blob/main/docs/en/tapdance.md) documentation
I kept running into the error ->
```
kmk.keys: Invalid key: HT
Traceback (most recent call last):
  File "code.py", line 32, in <module>
  File "kmk/keys.py", line 577, in argumented_key
  File "kmk/modules/tapdance.py", line 18, in __init__
TypeError: 'Key' object is not callable
```
After many hours of trying , I finally found a solution where i had to append the HoldTap Module into the keyboard variable 

code as follows -> 
```
from kmk.modules.tapdance import TapDance
from kmk.modules.holdtap import HoldTap

keyboard = KMKKeyboard()
tapdance = TapDance()
tapdance.tap_time = 750
HT = HoldTap()
keyboard.modules.append(HT)
keyboard.modules.append(tapdance)

.
.
.

MY_TD = KC.TD(KC.A, KC.B,tap_time=200)

keyboard.keymap = [
    [....,MY_TD ,......]
]

```
This addition into my code solved my problem and the function is now working as intended.
I believe this is a bug and either documentation or code needs to be updated

Thank you

gunz0blazin, 2024-07-30:
I ran into the same problem trying to get TD working on my custom firmware and your solution was what fixed it for me. I think you are correct, the example code needs to be updated to reflect that along with the documentation. I was banging my head into the wall for hours thank you!

xs5871, 2024-09-03:
Yes, tap dance shares code with hold-tap. Until very recently, that dependency on the hold-tap module being present was unavoidable. It was likely never documented, because they are usually used in conjunction.
That strong dependency is no longer strictly required.

closed 2024-09-03 by Crypto-Darth