KMKfw/kmk_firmware #1019: [BUG] cannot instantiate `macros`

miloth, 2024-08-18:
**Describe the bug**
`TypeError` when using the `macros` module.

**To Reproduce**
Simply follow the example for the [macros](https://github.com/KMKfw/kmk_firmware/blob/main/docs/en/macros.md) module:
```python
from kmk.kmk_keyboard import KMKKeyboard
from kmk.keys import KC
from kmk.modules.macros import Macros

keyboard = KMKKeyboard()
keyboard.modules.append(Macros())

GIVE_ME_THE_ANSWER = KC.MACRO("fortytwo")
```

**Expected behavior**
The last command should instantiate the macro, to be assigned to one of the keys in the keyboard mapping.

**Debug output**
I get this error:
```
104984 kmk.keys: Invalid key: MACRO
Traceback (most recent call last):
  File "code.py", line 19, in <module>
TypeError: 'Key' object is not callable
```

**Additional context**
N/A.

claycooper, 2024-08-18:
Can you post your whole `code.py` over at https://kmkfw.zulipchat.com/?

BOBO-dingdang, 2024-08-19:
You need to do this:
macros = Macros()
keyboard.modules.append(macros)

closed 2024-09-02 by claycooper